nodemcu/nodemcu-firmware #2625: ESP32: Touch module?

chilipeppr, 2019-01-23:
Anyone possibly working on a touch module? I'd love to be able to use the touch sensors on ESP32 from Lua. In particular would really like to be able to do from interrupts. Looking to control a 3D printed robot arm with an ESP32 on each stepper motor (there's 7 across the arm, so need 7 ESP32's for whole thing) with touch sensors to control each part of the arm for positioning and possible detection of getting bumped by a human.

devsaurus, 2019-01-29:
> Anyone possibly working on a touch module?

I'd interpret the silence as a No.
Are you looking into adding touch support for NodeMCU?

chilipeppr, 2019-01-30:
Yes. Much like the post I just did on the pulse counter, I would like to
write the touch module with a similar clean interface if nothing else is
out there on the interwebs. Just wanted to check with everyone first to not
do redundant work.

chilipeppr, 2019-07-23:
I have pretty much completed the Touch module. Just working on docs right now. Here's a preview of it fully working. https://youtu.be/HiTzed5MDb4

closed 2020-07-31 by stale[bot]