nodemcu/nodemcu-firmware #2626: [esp32] MQTT mutual auth

kevinresol, 2019-01-23:
Trying to get MQTT mutual authentication (client cert) work on esp32.

[Here is my attempt](https://github.com/kevinresol/nodemcu-firmware/commit/bea2c4813306d3c4ebe5a025df66a981120a12fb), trying to pass the PEM's to esp-mqtt.

And the lua call is:

```lua
mqttClient:connect(url, port, 1, 0, cacert, client_cert, private_key, on_ok, on_nok);
```

However, I am getting this error when connecting:

```
I (11446) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
E (11546) MQTT_CLIENT: Invalid MSG_TYPE response: 9, read_len: 0
I (11546) MQTT_CLIENT: Error MQTT Connected
I (11546) MQTT_CLIENT: Reconnect after 10000 ms
E (11556) MQTT_CLIENT: Client has not connected
```

`read_len` is zero meaning the response (9 in this case) is basically rubbish data in the buffer, we could ignore that.

[`select()`](https://github.com/espressif/esp-idf/blob/e2ae69f/components/tcp_transport/transport_ssl.c#L89) tells us the fd is ready to be read but [actually reading](https://github.com/espressif/esp-idf/blob/e2ae69f/components/tcp_transport/transport_ssl.c#L129) it yields `0` length. I guess that means the socket is closed. But I am not sure why.

I am not a C expert, I would like to have some guide on this problem. Thanks.

devsaurus, 2019-01-23:
I'm not sure about the overall status of mqtt over SSL, never tried that myself. Just to check basic things:
* you selected 'ESP-MQTT Configurations' --> 'Enable MQTT over SSL' in menuconfig?
* mqtt SSL connection works *without* client cert?

kevinresol, 2019-01-24:
Turns out it is working fine. I was using AWS IOT and there are some permission settings there disallowing connection. Fixing them fixed the problem.

Now I wonder what is the expected Lua API for this?

devsaurus, 2019-02-13:
Resolved with #2628, #2657.

closed 2019-02-13 by devsaurus