nodemcu/nodemcu-firmware #2634: ESP32: Anyone working on a pulse counter module?

chilipeppr, 2019-01-30:
### ESP32 Pulse Counter Module (New feature)
I'm considering writing a Lua module for the ESP32 Pulse Counter https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/pcnt.html

Before I do, I'm wondering if anybody has done any work on one so I don't reinvent the wheel? I need to  read steps from a stepper motor with perfect accuracy to record where the user jogged the stepper motor on a robot arm. I think the ESP32 pulse counter is perfect for this as it offloads the work from a Lua callback to the pulse counter hardware. The pulses from the stepper motor come in around 2us per step and the resolution of the ESP32 pulse counter is 12.5ns so I should be fine. For an alternate project I need to read the rotation of a spinning spindle from a hall effect pulse counter, thus the extra need for this module.

Doing this from a GPIO interrupt and counting from a Lua callback is not ideal as it puts too much load on the main ESP32 CPU, can become inaccurate, and callbacks can get missed. The hardware pulse counter, which I believe did not exist on ESP8266, but does on ESP32, looks to be an ideal solution here, but no module exists yet for Lua as far as I can find.

The docs look pretty clear from the URL above and I can mimic behavior from many of the other modules in terms of config and callback. 

The thoughts on how to call this would be:
```lua
pulse.config({
  unit=0, -- Defaults to 0. Use this value for all subsequent methods. ESP32 has 0 thru 7 units to count pulses on
  channel=0, -- Defaults to 0. Each unit has 2 independent channels 0 and 1 to count on
  gpioPulseInput=12, -- gpio for incoming pulses to be counted
  gpioCtrlInput=14, -- gpio for control input. See ctrlModeLow and ctrlModeHigh for what to do when control input is low or high.
  ctrlModeLow=pulse.PCNT_MODE_KEEP, -- Defaults to pulse.PCNT_MODE_KEEP (won’t change counter mode). What to do when gpioCtrlInput pin is low. Other options pulse.PCNT_MODE_REVERSE (invert counter mode. increase -> decrease, decrease -> increase), or pulse.PCNT_MODE_DISABLE (inhibit counter. counter value will not change in this condition)
  ctrlModeHigh=pulse.PCNT_MODE_DISABLE, -- Same as ctrlModeLow options, but for what happens on high pulse on gpioCtrlInput pin
  countModePosEdge=pulse.PCNT_COUNT_INC -- Default to pulse.PCNT_COUNT_INC (increase counter) . Other options pulse.PCNT_COUNT_DEC (decrease counter) and pulse.PCNT_COUNT_DIS (inhibit counting)
  countModeNegEdge=pulse.PCNT_COUNT_DIS, -- Same as countModePosEdge, but for negative edge
  ctrHighLimit=10, -- Defaults to 32767. Range int16 [-32768 : 32767]. Maximum counter value. Change later using pulse.callbackSetEventVal(unit, pulse.PCNT_EVT_H_LIM, value)
  ctrLowLimit=-10, -- Defaults to -32767. Range int16 [-32768 : 32767]. Minimum counter value. Change later using pulse.callbackSetEventVal(unit, pulse.PCNT_EVT_L_LIM, value)
  cb=myfunction, -- your Lua method to call on event. myfunction(unit, arg, event, threshold) will be called
  cbArg=0, -- your unique data that is passed back to you as an arg to your callback
  cbEvents={pulse.PCNT_EVT_L_LIM, pulse.PCNT_EVT_H_LIM, pulse.PCNT_EVT_ZERO, pulse.PCNT_EVT_THRES_0, pulse.PCNT_EVT_THRES_1}, -- provide any/all events you want the callback called on. Change later with pulse.callbackEventEnable(unit, event)
  ctrThresh0=1000, -- Defaults to 32767. Range int16 [-32768 : 32767]. Get callback when channel 0 gets a counter at or above this threshold. Change later using pulse.callbackSetEventVal(unit, pulse.PCNT_EVT_THRES_0, value)
  ctrThresh1=1000, -- Defaults to 32767. Range int16 [-32768 : 32767]. Get callback when channel 1 gets a counter at or above this threshold. Change later using pulse.callbackSetEventVal(unit, pulse.PCNT_EVT_THRES_1, value)
})

pulse.callbackEnable(0) -- enable callback on unit 0. Watch point event type. All enabled events share the same interrupt (one interrupt per pulse counter unit).
pulse.callbackDisable(0) -- disable callback on unit 0
pulse.callbackEventEnable(0, pulse.PCNT_EVT_L_LIM) -- on unit 0 watch point event: Minimum counter value
pulse.callbackEventEnable(0, pulse.PCNT_EVT_H_LIM ) -- on unit 0 watch point event: Maximum counter value
pulse.callbackEventEnable(0, pulse.PCNT_EVT_ZERO ) -- on unit 0 watch point event: counter value zero event
pulse.callbackEventDisable(0, pulse.PCNT_EVT_L_LIM) -- on unit 0 disable callback for Minimum counter value
pulse.callbackSetEventVal(0, pulse.PCNT_EVT_THRES_1 , 20) -- on unit 0 set the watch point event: threshold1 value to 20 to get a callback when 
pulse.callbackSetEventVal(0, pulse.PCNT_EVT_THRES_0, 1000) -- on unit 0 set the watch point event: threshold0 value to 1000 to get a callback when 
local val = pulse.callbackGetEventVal(0, pulse.PCNT_EVT_THRES_1) -- get value of threshold1 event callback trigger

-- get current pulse counter value
local pulseCtr = pulse.get(0) -- pass in which unit to get value from

-- pause counting
pulse.pause(0) -- pass in unit

-- resume counting
pulse.resume(0) -- pass in unit

-- clear counting
pulse.clear(0) -- pass in unit

-- change the pin "on the fly" to count pulses on and to control counting
pulse.setPin(0, 0, 23, 24) -- change unit 0, channel 0 to gpioPulseInput=23 and gpioCtrlInput=24

-- remove pin altogether "on the fly"
pulse.setPin(0, 0, pulse.PCNT_PIN_NOT_USED, pulse.PCNT_PIN_NOT_USED)

-- set a filter value to ignore pulses for an amount of clock cycles
pulse.setFilter(0, 10) -- ignore 10 clock cycles = 125ns. each cycle is 12.5ns based on 80Mhz APB clock. filter_val is a 10-bit value, so the maximum filter_val should be limited to 1023.
pulse.filterEnable(0) -- turn on filter
pulse.filterDisable(0) -- turn off filter

-- check current filter value setting
local filterVal = pulse.getFilter(0) -- get filter clock cycle ignore value for unit 0

-- disable pulse counting on a unit
pulse.disable(0) -- which unit to disable
`

devsaurus, 2019-02-03:
Thanks for proposing this new module @chilipeppr!

I first had read through the IDF and technical reference manual to better understand the example code above. Especially the concept of units and channels is not obvious from IDF's API. Took me some time to grasp that `pcnt_unit_config()` doesn't configure a complete unit, but deals only with one of its two channels.
The Lua API for `pulse.config()` should be clearer in this respect:
* Handle all resources of a unit
    Enable the user to configure of both channels (
* Obvious assignment of configuration items to resources
    Separation between channel- and unit-related items

IDF's `pcnt_unit_config()` obviously doesn't provide that, but it should be feasible to encapsulate this in the C layer. Something in the direction of
```lua
pulse.config(unit, {
  channel0 = {
    gpioPulseInput = ...,
    gpioCtrlInput = ...,
    ctrlModeLow = ...,
    ctrlModeHigh = ...,
    countModePosEdge = ...,
    countModeNegEdge = ...,
  },
  channel1 = {
    -- same as above
  },
  -- the unit-related items
})
```

The `cbArg` parameter is not needed IMO. Passing complex context information/objects to a callback works for both global and local variables, while the latter are treated as upvals in this case. Implementing `cbArg` would just wrap that with a manually programmed layer in the C module.
```lua
function init_pulse_counter(unit, context)
  pulse.config({
    cb = function (unit, event, threshold) my_callback(unit, context, event, threshold) end
  })
end
```

chilipeppr, 2019-02-03:
Ahh. Interesting. I didn't realize you had to configure both channels. I read through the docs over and over as well and that was confusing me. Without actually getting to play with the pulse counting, it's hard for me to really understand how it works. You've shed light on that in your description and I like your config approach better. I also agree you don't need a cbArg. I was just trying to expose what the IDF exposes, but in a Lua world it isn't really needed. 

Now, one thing that I really don't understand then is the methods the IDF gives you to dynamically change stuff on the fly, you would have to specify unit and channel. The only method that accepts a unit and channel that I can see is:
```lua
-- change the pin "on the fly" to count pulses on and to control counting
pulse.setPin(0, 0, 23, 24) -- change unit 0, channel 0 to gpioPulseInput=23 and gpioCtrlInput=24
```
I would assume each channel has its own pulse counter, but that can't be since you can only get one pulse counter per unit. I suppose both channels just modify the main unit pulse counter.
```lua
-- get current pulse counter value
local pulseCtr = pulse.get(0) -- pass in which unit to get value from
```
Ok, well, I'll see if I get time for all of this. Writing Lua is much more pleasurable. I find the workflow of compiling the C code, uploading to the ESP32, then running test Lua code to test the C code is a ton of work and it's slow.

devsaurus, 2019-02-11:
> I would assume each channel has its own pulse counter

I suspected this as well in the beginning. But it appears that two paired channels operate on a single counter resource. The architecture diagram in the technical reference manual also suggests this view, and IDF's API is consistent with that (of course). E.g. all 4 filters for the channels are configured only once per counter etc.

> The only method that accepts a unit and channel that I can see is `pulse.setPin()`

Yes, pins are dedicated resources for channels, not counters.

devsaurus, 2019-07-21:
PR merged

closed 2019-07-21 by devsaurus