nodemcu/nodemcu-firmware #2639: [ESP32] Bluetooth Serial support

kevinresol, 2019-01-31:
### Missing feature
Bluetooth serial aka classic bluetooth (instead of BLE) on ESP32

### Justification
Classic bluetooth itself is useful for streaming, e.g. file transfer. 

### Workarounds
Not really any. Might be possible to upload files to a server and use wifi to download. But it is much more complex.

nwf, 2019-01-31:
Please see #1010.  Are you in a position to work on BT SPP patches?  I'm sure we'd take them, but I don't think we have the resources to produce them.

kevinresol, 2019-01-31:
My knowledge of lua/c is scarce. My most fear is about memory management. I will take some time to learn them. I will be back but closing for now. Thanks!

closed 2019-01-31 by kevinresol

Alfrederson, 2019-06-10:
> ### Reference
> https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/src/BluetoothSerial.cpp

Adapting this file kind of worked for me (I'll post source code late), but there are some problems that keep it from working 100% as if it was just typing commands into a serial terminal. E.g: calling file.open() crashes the interpreter and I haven't figured out why yet. I don't know if it's because of something I'm doing wrong or due to some internal features of the interpreter.