nodemcu/nodemcu-firmware #2640: DNS resolution failure when using tls.socket:dns()

NicolSpies, 2019-01-31:
### Expected behavior
It is expected that tls.socket:dns() provides a DNS resolution for a hostname.
### Actual behavior
Callback function never fires indicating no resolving. Resolving of domain was confirmed with `net.dns.resolve(host, function(sk, ip))`
### Test code
```Lua
sk = tls.createConnection()
sk:dns("google.com", function(conn, ip) print(ip) end)
```
### NodeMCU version
```Lua
 NodeMCU 2.2.0.0 built with Docker provided by frightanic.com
	branch: dev
	commit: 0e89fb24bae3c3dcd446b64f9bbc6d9581c08632
	SSL: true
	Build type: float
	LFS: Size: 0x10000
	modules: adxl345,cron,file,gpio,http,i2c,net,node,rtcmem,rtctime,sjson,tls,tmr,uart,wifi
 build created on 2019-01-23 10:59
 powered by Lua 5.1.4 on SDK 2.2.1(6ab97e9)
```
### Hardware
Standard ESP12 hardware configuration.

nwf, 2019-01-31:
Why do we have separate `net.dns` and `tls.dns` functions, anyway?

NicolSpies, 2019-01-31:
I believe it might be related to the fact that when you create a `tls.createConnection` the tls.socket sub module returned have a number of methods of which dns is one of them.

nwf, 2019-01-31:
That's not the right kind of answer to a "why" question.

marcelstoer, 2019-02-01:
👍 for dumping `tls.dns` unless @djphoenix who wrote that (or anyone else) comes forward with good reasons for keeping it.

commit 06367cb7 (2019-02-02): Tell the truth about DNS in TLS module (#2643)

Purge the never-functional :dns() method, but document the DNS callback
in :on().

Fixes #2640
closed 2019-02-02 by marcelstoer