nodemcu/nodemcu-firmware #2647: Http server doesn't return data: Connection reset by peer

libscott, 2019-02-05:
### Expected behavior

Using the example in http-example.lua, client will receive "Hello, World!" 200 response

### Actual behavior

Server:

    > +R    GET     /       20792
    +B      0       20696
    +B      nil     19848

Client:

    > curl http://192.168.0.122/
    curl: (56) Recv failure: Connection reset by peer

### Test code

Exactly the code here: https://github.com/nodemcu/nodemcu-firmware/blob/e87aef8328beea4b5f59ae633bfaa6ddaf24cf7f/lua_modules/http/http-example.lua

### NodeMCU version

```
NodeMCU custom build by frightanic.com
        branch: dev
        commit: e87aef8328beea4b5f59ae633bfaa6ddaf24cf7f
        SSL: false
        modules: adc,bit,ds18b20,file,gpio,i2c,net,node,tmr,wifi
 build created on 2019-02-02 18:58
 powered by Lua 5.1.4 on SDK 2.2.1(6ab97e9)
```

### Hardware

Lolin NodeMCU Lua V2

> print(node.info())
2       2       0       582959  1458280 4096    2       40000000

libscott, 2019-02-05:
I believe this might be related to #730, ie, close happening before all data has been flushed.

nwf, 2019-02-05:
@libscott The Espressif-provided primitives are poor for abstraction.  I think the correct answer here, tho', is to adopt @TerryE's fifo trick around the req's `:send` and re-expose a "fifo has room" callback and make the server's `:finish` wait for the fifo to drain.  You can see something of the dance I mean at https://github.com/nodemcu/nodemcu-firmware/blob/dev/lua_examples/telnet/telnet.lua or https://github.com/AcmeTensorToys/esp-lua_core/blob/master/net/fifosock.lua but this would need additional effort for `:finish()`.  Doing that would be a worthwhile exercise for the NodeMCU ecosystem, I think.

libscott, 2019-02-06:
@nwf the example I provided also leaks memory, any ideas?

nwf, 2019-02-06:
@libscott Not a clue.  Are you sure it's a leak proper and not just the TCP connection in TIME_WAIT state or such transients that will go away with time or heap pressure?  If it *is* a real leak, there's probably a knot tied through the lua registry, possibly between the socket and its callbacks; you might try clearing them out with `s:on("sent", nil)` and such.

libscott, 2019-02-06:
Nice! Your idea of unregistering the event handlers seemed to do the trick. From losing ~800b per request to ~100b and GC eventually recoups it all back. Thanks for the fast response.

```
req.conn:on("sent", function(s)                                    
                        s:close()                                  
                        s:on("sent", nil)                          
                        s:on("receive", nil)                       
                        s:on("disconnection", nil)                 
                    end)                                           
```

closed 2019-02-06 by marcelstoer

libscott, 2019-02-06:
@marcelstoer why was this closed? For my purposes I have found a workaround but the issues remain.