nodemcu/nodemcu-firmware #2648: Please cleanup trailing spaces

c-e-p-x-u-o, 2019-02-06:
There are a lot of trailing spaces that reduces readability and should be cleaned up.

c-e-p-x-u-o, 2019-02-06:
fixed via #2649

ziggurat29, 2019-02-12:
and in some cases, trailing spaces can break things.  e.g. trailing spaces after a backslash that is a line continuation character in a multi-line #define, at least if the compiler strictly adheres to the standard.  E.g. C++ standard section 2.2:

> "Each instance of a backslash character () **immediately** followed by a new-line character is deleted,
> splicing physical source lines to form logical source lines."

(emphasis added)

TerryE, 2019-02-15:
In principle, doing this and any tab expansion cleanup is good house keeping and it makes sense to do this. 

However, the converse to @ziggurat29's point is also true: it is possible that in rare cases trailing blanks might have semantic significance and should not be removed.  So we should just present this as a PR without at least doing the static analysis and reviewing this first.

PR #2659 (2019-02-17): trailing spaces cleanup
Fixes #2648

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

There are a lot of trailing spaces that reduces readability and should be cleaned up.
closed 2019-02-19 by TerryE