nodemcu/nodemcu-firmware #2651: add support for dhcp ntp server

c-e-p-x-u-o, 2019-02-06:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and **DELETE THE REST**.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

### Actual behavior

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
-- add code here
```
### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.

### Hardware
Describe which ESP8266 device you use and document any special hardware setup 
required to reproduce the problem.

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

### Justification
Tell us why you would like to see this feature added.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

8<------------------------ END FEATURE REQUEST --------------------------------

nwf, 2019-02-06:
See #1010.  Also you should be more clear what you mean; NTP does not run over DHCP, so either you mean for setting the DHCP option identifying the NTP host in DHCP offers (relatively easy) or for the ESP8266 to be a NTP server (way harder).  But for either one, see #1010 or contact one of us out of band for contracting rates.

c-e-p-x-u-o, 2019-02-07:
I mean dhcp option 42.
I have the working patch, that I'll PR tomorrow.

marcelstoer, 2019-02-07:
@nwf the author included the patch for this by mistake in the (now closed) PR #2649.

nwf, 2019-02-07:
Sorry, the issue text made it sound like this was another request for development effort.  I look forward to the PR.

PR #2709 (2019-07-27): support for dhcp ntp server (option 42) reloaded
Fixes #2651.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`. (why do I get an old template here with wrong path /en)

support for ntp server offered via dhcp (option 42)

since #2660 is unchanged for over one month I try this one.

I t [... cut]
closed 2019-07-27 by marcelstoer