nodemcu/nodemcu-firmware #2652: Not flashing at all

UrafikiTheFluff, 2019-02-09:
ok so i got the Ämica"one and i got 4 others from a differnet company they all have the cp2012 driver and stuff i have used the "nodemcu firmware programmer"and ive used audrinoide i just dont know what the heck to do i need some help can someone lead me to the right direction please a actual youtube help video im just not getting like uk the wifi thing were i get on my android phone it still shows like ""ESP_212374" this is becoming annoying please help me

nwf, 2019-02-09:
Please see #1010; this is not a general nodemcu support forum.

UrafikiTheFluff, 2019-02-09:
where do i get help then?

nwf, 2019-02-09:
As per #1010, please see our support page, https://nodemcu.readthedocs.io/en/latest/support/

closed 2019-02-10 by marcelstoer