nodemcu/nodemcu-firmware #2656: LFS setup example in docs

cwrseck, 2019-02-10:
### Expected behavior
An example in the documentation showing how to set up an LFS system.

### Actual behavior
The information is available, but scattered.

I've been setting up an LFS esp8266, and found bits of the LFS docs
confusing; all the information is there, but it would help if there
were a simple sequence of commands to demonstrate LFS use.  The following
summary might be worth adding to the docs, though I'm not sure where.

8<------8<------8<------8<------8<------8<------8<------8<------8<------

To use LFS, build a version of NodeMCU with LUA_FLASH_STORE set in
user_config.h, and load it on the esp8266 in the usual way (whatever
that is for your set up).

Then build an LFS file system, which becomes a separate file on the
esp8266's SPIFFS.  An LFS file system can be built in several ways;
one of the easiest is to use "luac.cross -f -o lfs.img *lua" on the
host machine.  The file "_init.lua" from lua_examples/lfs should be
included in the image, since it's the easiest way of registering the
LFS modules.  The lfs.img file can then be downloaded to the esp8266
just like any other file.

The next step is to tell the esp8266 that the LFS file system exists.
This is done with eg. node.flashreload("lfs.img"), followed by
node.flashindex("_init") to register the modules; logging into the
esp8266 and running the following commands gives an overview of the
sequence. (NB. no error correction is used.)

>
> node.flashreload("lfs.img")
> print(LFS)
nil
> node.flashindex("_init")()
> print(LFS)
table: 3fff06e0
> print(LFS._list)
table: 3fff0728
> for k,v in pairs(LFS._list)  do print(k,v) end
1    dummy_strings
2    _init
3    main
> LFS.main()
LFS main() module
> 

After that, it's a question of setting up the boot process to check for
the existence of an LFS image and run whichever module is required.
Since the LFS table has been registered by running _init, running an
LFS module is simple a matter of eg: LFS.main()

8<------8<------8<------8<------8<------8<------8<------8<------8<------

C W Rose

nwf, 2019-02-10:
Seems like a good idea; please open a PR.

HHHartmann, 2019-02-10:
This is definitely a good idea to have such LFS quichstart. Also not starting at the ABCs but assuming that the user knows his/her way around already a bit.

Please call it "LFS image" and not "LFS file system" as that's what it is called elsewhere.
Also note that ```node.flashreload("lfs.img")``` will reboot the system on success (and return error message on failure which you might print)
Maybe also briefly describe what the steps do and why the result is as it is.

In the end a working "Hello World" would be nice. Someone asked for that a while ago already.
Just have a modifiled _init.lua and a matching init.lua calling the HelloWorld() in the _init.lua (or maybe have the HelloWorld in a separate .lua file.)
A Hint to ```require``` and ```loadfile``` as alternatives to ```LFS.``` might also help.

For many users it might not be the easiest way to use luac.cross to get an LFS image as they don't have a build system at hand.
So maybe rather refer to the online service maintained by @TerryE 

"build an LFS file system, which becomes a separate file on the esp8266's SPIFFS"
This might sound like the LFS image is built on the ESP, maybe make it clear that the image has to be loaded on the SPIFFS in a separate step.

nwf, 2019-02-10:
"build an LFS file system, which becomes a separate file on the esp8266's SPIFFS" also makes it sound like the LFS permanently lives in SPIFFS, which it does not.

ziggurat29, 2019-02-10:
if you're going to mention _init.lua, don't forget about dummy_strings.lua.

The LUA_FLASH_STORE is important if you're building the firmware yourself, but that detail is handled automatically if you're using the online build system (which as I understand it is the majority of users).

@nfw makes a good point that clarifying that the built lfs image that is copied to spiffs is simply a staging copy.  Once node.flashreload("lfs.img") has happened, the file on spiffs can be deleted -- it doesn't serve any further purpose.

devsaurus, 2019-02-11:
For sure a good idea!
As an "LFS beginner" I would even see this like a tutorial which describes different use cases. Starting from the most simple to more complex ones.
- How can I put HelloWorld.lua into LFS and call it afterwards?
- How can I move a Lua module to LFS and how it's used from there?
- Further tweaks and hooks in `_init.lua` explained.

I found `lua_examples/lfs/_init.lua` to support all of my basic needs out of the box. Guess it should be promoted right from the start.

These are very *basic* items, so they should return instant success for a beginner, bringing him up to speed with LFS :smiley:

cwrseck, 2019-02-12:
Unfortunately the esp8266 terminal session used '>' as a prompt,
which GitHub takes as a quote marker.  To avoid this, the code
block should have been enclosed in triple backticks.

Again, it's unclear after the reformatting but LFS.main() calls
an actual file, main.lua, which has the contents:
print("LFS main() module")

The example is intended to be trivial, hence the omission of error
checking; there's a complete explanation to be found in the current
documentation, but an overview of the whole sequence is something I
would have found useful when starting to use LFS.

Some comment of the form: "Once node.flashreload() has loaded the LFS
image into flash memory the original file (in SPIFFS) is no longer used,
and can be deleted." should probably be added somewhere, to emphasize
LFS' use of flash memory.

C W Rose

TerryE, 2019-02-15:
We've already got the [LFS quickstart stuff] (https://nodemcu.readthedocs.io/en/dev/getting-started/#compile-lua-into-lfs-image) in the Lede article and the [LFS white paper](https://nodemcu.readthedocs.io/en/dev/lfs/). But certainly in the case of the latter, it is probably too heavy going for the average IoT Lua developer who isn't really interested in how this magic works, but more just a short explanation of what LFS does -- which to allow you to run Lua code out of Flash, so you can run large Lua Apps and still have all RAM available for data -- and a few worked examples of how best to use LFS.  In this visual age, this would probably be best done as a Youtube video, and a series of blog posts, IMO

cwrseck, 2019-02-16:
As I said. the information is available, but scattered.  Having just revised my ESP8266 setup,
the notes in the original post were what I needed to know to get LFS up and running; having
that simple example in one place would have made the update a lot easier.  Scattered YouTube
and blog posts wouldn't have been especially helpful.

C W Rose

TerryE, 2019-02-16:
@cwrseck 

> **_Scattered_** YouTube and blog posts wouldn't have been especially helpful.

\[My ital] I agree that many developers don't have the patience to mine Internet resources and often Google search rankings make good material hard to find.  

I and the other contributors don't earn a penny doing this work.  For me at least, this is just a way of continuing to do what I have always loved doing, and at the the same time giving to the community.  We all welcome the constructive contribution of other developers.  OK few might have the in depth knowledge of doing embedded IoT development in C and the internals of the Lua engine, but most can at least contribute aspects of their learnings that will help others follow the same path, and if developers repay the community by sharing their experience and learnings, then at least we have to option of aggregating the best of these or at least reference these for onward reading in our FAQs. :smile:

jmd13391, 2019-02-16:
HI All, I'm not sure this would be of any value to anyone but in the off chance it is...
I whipped-up a quick & dirty procedure for Windows that I follow to build LFS Enabled NodeMCU Firmware. This is wet paint (draft). Comments welcome.

[WIN_ESP8266_Firmware-LFS_Build_Steps.txt](https://github.com/nodemcu/nodemcu-firmware/files/2872143/WIN_ESP8266_Firmware-LFS_Build_Steps.txt)

TerryE, 2019-02-18:
@jmd13391 Joe, Why not create this as an MD file rather than flat text and put it in your gist folder on GitHub?

@cwrseck can you please take a look at `docs/getting-started.md` which is the Getting Started section in the documentation and integrate your suggestion into this as a PR?

jmd13391, 2019-02-18:
@TerryE Will do! I guess this will force me to read-up on how to create a md file. 😁

TerryE, 2019-02-19:
Look at the styling link in the new comment box, and view some of the other MD files in raw mode. MD is a pretty simple way of generating rich text formatted posts that are readable.🙂

cwrseck, 2019-02-21:
As always, git was a knock-down drag-out fight.  However, the forked repository cwrseck/nodemcu-firmware now contains  an updated getting-started.md.  I can't find a way of actually running a pull
request, since that needs access to the original repository.

C W Rose

IDispose, 2019-02-26:
@cwrseck 

Thank you very much for including n00b steps in the documentation. I struggled with the docs and your update and examples help clear up the confusion. 

One thing I am not clear on is, your example walks thru the process of uploading the `.img` file, reloading the flash and calling `_init` via "command line", but what happens when I manually reset the board via the reset button? 

From the docs

> After that, it's a question of setting up the ESP8266 boot process to check for the existence of an LFS image and run whichever module is required. Once the LFS module table has been registered by running lua_examples/lfs/_init.lua , running an LFS module is simple a matter of eg: LFS.main()

In my `init.lua` am I expected to check for existence of LFS image using `file.exists()` and also execute `node.flashindex("_init")()` before I execute `LFS .main()`?

Thank you again for updating the docs.

cwrseck, 2019-02-26:
Reboot works the same way with and without LFS; it runs the file "init.lua"
(or possibly "init.lc", I haven't tried that) from SPIFFS.  To run LFS stuff,
you need to check that the LFS file system exists and that it contains the
main() program you want to run.

I'm not clear as to the difference, as far as dofile() is concerned,
between main.lc and main.lua.  Generally I use main.lc, since compiling
large files runs out of memory.

C W Rose

jmd13391, 2019-02-26:
@IDispose Have you looked over the step by step text file I posted a few comments above? Does any of it help clarify the process for you?

IDispose, 2019-02-26:
@jmd13391 Yes, I checked the file and thanks for including the instructions. Your list also has examples using `nodemcu-tool` to directly execute code via "command line". I wasn't clear on what happens when I hit the `Rst` button and let the board recycle.

@cwrseck Thanks for the additional inputs. Is the following correct?

To use LFS, 

- Create folder named `myImage` 
- Copy all the .lua files that need to be in the LFS image into `myImage`
- Download and copy `lua_examples/lfs/_init.lua` into `myImage`
- Create the '`LFS image` using  either `luac.cross -f -o lfs.img *lua` or cloud service
- Upload `lfs.img` to the board using `nodemcu-tool`
- Execute `node.flashreload("lfs.img")` from "command line"
- Create `init.lua` with this code
- 

```
if(LFS) == false and file.exists('lfs.img')
  node.flashindex("_init")()
end
LFS.main()
``` 

- Upload `init.lua` using `nodemcu-tool`

Now every time the board reboots, it will use LFS

TerryE, 2019-02-26:
Hi guys, this thread really falls outside the scope of an issue, but I am happy to let this run for a little because I'd like you guys to take these learnings and update the intro documentation. :smile: 

First the use of an `lc` vs `lua` extension is a convention and has no semantic significance to the Lua VM: all is cares about is whether the file has a valid binary header (**\<ESC>Lua**) or not; if it has then the binary loader is executed; if not then the file is compiled and loaded.  By default the startup sequence attempts to execute `init.lua` from SPIFFS, but there is nothing to stop you renaming a compiled file "init.lua": the Lua VM will still load and execute it at startup.

There is a build define `LUA_INIT_STRING` which if define overrides this default of  `dofile "init.lua"`.  Here is the commented out example, though you are entirely free to replace this with your own version.
```C
#define LUA_INIT_STRING "pcall(function() node.flashindex'_init'() end)"
```
If LFS is enabled and module `_init` exists in the store then `node.flashindex'_init'()` will execute this.  If it doesn't then executing this statement will throw an error.  By wrapping it in a `pcall()`, the function will always succeed, albeit in this thrown case returning `false,<error message>`.  What this will do if you don't have an LFS `_init` module which also starts your application is to leave the Lua VM waiting for interactive input in UART0.

cwrseck, 2019-03-13:
Looking at the LFS build process, eg: luac.cross -f -o lfs.img *lua, it's not clear whether or not the .lua files are simultaneously compiled to .lc.  The LFS._index method just gives the module name, not the extension.  It's probably worth adding a comment to the docs clarifying the actual behaviour.

C W Rose

TerryE, 2019-03-23:
This has been a useful to-and-fro, but I will close this, unless someone proposes a PR rolling this up.

nwf, 2019-04-09:
@poorandunlucky: Write one up and open a PR?

poorandunlucky, 2019-04-09:
@nwf I was considering essentially taking notes, making a clean text out of that, and putting a little example/tutorial with it, but I'm not sure if there's an expected format?  I guess I'll see what I can do...  I would just hate to put a lot of work into it and it to go to waste...

I guess I'm there right now, might as well...  I think I'll clean-up what I'm working on right now, and when I'm ready to import stuff to LFS I'll come back, and try to draft something as I'm doing it...

No promises, but I might try...

TerryE, 2019-04-09:
@poorandunlucky Patrick, I wrote the [LFS whitepaper](https://nodemcu.readthedocs.io/en/dev/lfs/) very much from the perspective of a core developer trying to identify an architectural approach to extend the capabilities of the NodeMCU firmware.  This perhaps too internals focused and most IoT developers don't really care about the magic just the tl;dr: 

_You can create a Lua code shareable image that the Lua runtime can load into, and execute directly from, flash memory; if you do this then you can use the RAM just for variable / data storage, and this effective increases the ESP8266 program capacity by maybe 2-4 ×_ 

OK there are some inconveniences and constraints, but quite honestly I don't piss around with executing Lua out of SPIFFS anymore as it is so quick to recreate a new LFS image and load it OTA.  The major hassle for WinX developers is that ATM they need to compile `luac.cross` themselves; this is a 10 minute task, but still one that most Windows developers find difficult.  However we are looking at a cloud builder option to be able to get the image (a Windows exe file) from a cloud resource for those that don't fancy Downloading one of the free C development environments onto their PC.

As far as coding style goes, I used to do all sorts of tricks to Just-in-Time load code and data to keep the RAM footprint small.  Now I find that there is no point;  it's just easier and simpler to keep the coding style simple and readable.

cwrseck, 2019-04-10:
poorandunlucky: in updating the docs it might be worth starting with
```
https://github.com/cwrseck/nodemcu-firmware/blob/master/docs/en/getting-started.md
```

It didn't make the current release, but if you can generate a pull request
(I couldn't) then I think a good few users would find the information helpful.

There are several other LFS questions worth answering in this thread, if you
can work out what the answers are (I couldn't).

Will

cwrseck, 2019-04-14:
Reference my comment of March 13, the answer is no; luac.cross of 20190405
segfaults when given an .lc file to include in an lfs image.  I don't know
if this is a bug or a feature.

C W Rose

TerryE, 2019-04-14:
@cwrseck Will, it would help us to respond constructively to your comments if you could follow the guidelines detailed in #1010.  Specifically This should be a separate issue; please use MD and provide a repeatable test case, for example:
```Bash
> cd $NODEMCU-FIRMWARE
> ./luac.cross -o /tmp/_init.lc lua_examples/lfs/_init.lua
> ./luac.cross -o /tmp/lfs.img  lua_examples/lfs/dummy_strings.lua /tmp/_init.lc
Segmentation fault (core dumped)
```
Thanks.

I can't think of any reason why you would want to include a pre-compiles Lua file rather then just Lua sources, but off the top of my head, I can't see why this should fail.  The workaround is simply to use the source version as input.  However I will add this to my TODO list and gather this up next time I do some updates to luac.cross.  If you want to track it, then you will need to raise this as its own issue.

ziggurat29, 2019-04-15:
@TerryE if it helps at all, I did reproduce the crash when doing a luac.cross with a pre-compiled Lua chunk.  FWIW, I found the crash to fundamentally be at:
`lauxlib.c:738`
in the `getF()` implementation.  This is because the `size` parameter is NULL, and is being deref'ed.
Going up the stack, this is in turn because at:
`lundump.c:238 `
in the `LoadFunction()` implementation there is code ` if (luaZ_direct_mode(S->Z)) l_setbit((f)->marked, READONLYBIT);`.  The `luaZ_direct_mode` is a macro that in turn uses another macro `luaZ_get_base_address` which performs `(zio)->reader()`.  `reader` in this case is pointing to the getF() function, and it explicitly passes NULL for the first and last parameter (the first parameter is unused in getF(), but the last one is the 'size' param mentioned before, hence the crash).

I also built the stock Lua 5.1.4 code, and observed this crashing did _not_ occur.  Looking at that same LoadFunction() code, I notice that the line involving `luaZ_direct_mode` is _not_ present, so I assume it is something that was added for eLua.

I'm not familiar enough with the eLua codebase to make any suggestions for a fix, but I thought this diagnostic info might be useful for someone who is more familiar.

I do agree that a separate bug filing should be made for this, but I wanted to at least record this info here so it could be transferred to such filing if this info is considered useful.

TerryE, 2019-04-15:
@ziggurat29, this one got me intreged as well and I did the same exercise as you and came to the same conclusion, though I didn't write it up because this issue is a documentation one. I suspect that the underlying issue is a subtle cross compile one causing the undump to fail.

TerryE, 2019-04-15:
Note that I have moved the content related to #2727 to that issue.  This includes content from myself, @cwrseck  and @ziggurat29

poorandunlucky, 2019-04-17:
Well, as I was invited to do so, I created a sort of "Quick Start Guide", here's a [link to the main file](https://github.com/poorandunlucky/lfs-qsg/blob/master/lfs-quickstart.md)...  I'm not sure how to implement it in the tree to issue a pull request, though...  I guess this is more a request for comments..?

PR #2807 (2019-06-21): Update LFS introduction
Fixes  #2656
- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

Continuation of PR #2795. Added only little changes to original @cwrseck work.
closed 2020-04-25 by stale[bot]