nodemcu/nodemcu-firmware #2662: Further deprecation of mqtt's autoreconnet

devsaurus, 2019-02-11:
There's still the deprecation note for the autoreconnect feature in `mqtt`:
https://github.com/nodemcu/nodemcu-firmware/blob/30744afd24ab861ba1a8ff3cf99a4c3e258a2221/app/modules/mqtt.c#L1333
I stumbled upon this during the [recent API discussion](https://github.com/nodemcu/nodemcu-firmware/pull/2628#issuecomment-458332617) for `mqtt` on ESP32.

How about next steps:
1. Print the warning not only when `autoreconnect` is enabled (status quo), but in any case when this parameter is specified.
    The parameter is optional, thus scripts can be adapted right away.
2. Wait for next master drop.
3. Finally remove `autoreconnect` from the API.

I'd set up respective PRs once we have a conclusion.

marcelstoer, 2019-02-11:
👍

nwf, 2019-02-12:
SGTM. :+1:

nwf, 2019-08-16:
mqtt reconnect removed by #2822.

closed 2019-08-16 by nwf