nodemcu/nodemcu-firmware #2663: Please add support for tftp client.

c-e-p-x-u-o, 2019-02-11:
Please add support for tftp client.
I'll try to make a pr, but not sure.

The good one that I know is tftp-hpa https://git.kernel.org/pub/scm/network/tftp/tftp-hpa.git

Also it will be nice to receive server's ip over dhcp.

TerryE, 2019-02-12:
> Please add support for tftp client

We already have a Lua module to a minimal FTP server in our lua_examples folder.  I can't see any advantage in slimming this down to TFTP, but you are welcome to try this for yourself.

>  it will be nice to receive server's ip over dhcp.

This is not an ESP function, but rather a DHCP one.  In station mode the DHCP look is a standard DHCP service function.  In AP mode, the ESP acts as the gateway.  IN both cases the IP address is available.

c-e-p-x-u-o, 2019-02-12:
How "FTP server" is relate to "tftp client"?
Please reopen!

c-e-p-x-u-o, 2019-02-12:
tftp has nothing common with ftp!

c-e-p-x-u-o, 2019-02-12:
http client is ESP function: https://nodemcu.readthedocs.io/en/master/en/modules/http/
mqtt client is ESP function: https://nodemcu.readthedocs.io/en/master/en/modules/mqtt/
tftp client is also ESP function!

nwf, 2019-02-12:
@c-e-p-x-u-o I agree a TFTP client could be of some utility for NodeMCU, but as per #1010, just open a PR with your contribution when it's ready.

c-e-p-x-u-o, 2019-02-12:
Issue should be left opened as it's not fixed.

c-e-p-x-u-o, 2019-02-12:
I'd like to updated lua scripts from external TFTP server.
The next step is OTA over TFTP offered by DHCP.

nwf, 2019-02-12:
@c-e-p-x-u-o For the love of my inbox, type longer comments rather than splitting thoughts across several.  I understand why you might want TFTP support, but none of us are, AFAIK, likely to work on it; we'll review a PR when you open one.

c-e-p-x-u-o, 2019-02-12:
Sorry.

TerryE, 2019-02-12:
@c-e-p-x-u-o, as Nathaniel suggests, please follow our contribution guidelines and GitHub issues etiquette when posting to an issue.  If you want to use an IRC paradigm then we also have a gitter thread.  I've merged some of your fragment posts. 

I reopened the issue temporarily since as you point out this is about _client_ not server functionality.  It would be very straightforward implementing this as an LFS-style Lua module.  See my `lua_examples/lfs/HTTP_OTA.lua` as a starting point.  Feel free to raise a PR when you've got this working.

marcelstoer, 2019-02-12:
> I'll try to make a pr

Great, that's the right spirit!

Thanks for following [our guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/master/CONTRIBUTING.md) by first creating an issue before raising a PR. We appreciate that.

TerryE, 2019-02-15:
@532910 Sergio, TFTP is a nice keep-it-simple protocal for local LAN and Wifi deployments and as I said above, it should be straightforward to implement this as an LFS compatible Lua module.  As Marcel says: have a go and upload it to your own github fork.  We will constructively support you in this :smile:

532910, 2020-02-12:
Dear bot, please don't close it, I'll try to implement it, I promiss (:

closed 2021-03-19 by stale[bot]