nodemcu/nodemcu-firmware #2664: ntp timezone over dhcp tcode pcode options 100 101

c-e-p-x-u-o, 2019-02-12:
Please add support for ntp timezone over dhcp tcode/pcode options 100 101
I'll try to make a pr, but not sure I can implement it.

Related RFC: https://tools.ietf.org/html/rfc4833

pcode -> tcode in lua:
https://github.com/openwrt/luci/blob/master/modules/luci-base/luasrc/sys/zoneinfo/tzdata.lua

pjsg, 2019-02-12:
I think that it would make sense to save the option 100 and 101 values, and then provide a LUA api to be able to get those values. The actual timezone conversion should probably be done in LUA code.... I have some code that interprets a compiled tzinfo file.....

c-e-p-x-u-o, 2019-02-12:
agree!

marcelstoer, 2019-02-12:
> [I have some code](https://github.com/nodemcu/nodemcu-firmware/tree/master/lua_examples/timezone) that interprets a compiled tzinfo file

I wonder why have this in `/lua_examples` rather than in `/lua_modules`?

pjsg, 2019-02-12:
Probably because lua_modules didn't exist at that time.

marcelstoer, 2019-02-12:
More OT, sorry: that folder goes way back to Dec 2014 😉

532910, 2020-02-09:
Not sure if it's clear for me how to implement it.

HHHartmann, 2021-02-13:
Maybe we shoud at least move the timezone example to modules then

closed 2022-04-30 by stale[bot]