nodemcu/nodemcu-firmware #2666: bluetooth error

funwun, 2019-02-13:
### Expected behavior
when I use bthci in esp32s module, always get the error code: 31

### Actual behavior

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
bthci.reset(function(err) print(err or "Ok!") end)
bthci.scan.enable(1, function(err) print(err or "Ok!") end)
-- 31
```
### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.
last of dev-esp32 branch, and build with docker

sdkconfig
```
CONFIG_BT_ENABLED=y
CONFIG_LUA_MODULE_BTHCI=y
```

### Hardware
ESP32S module

devsaurus, 2019-02-13:
I can confirm error 31 with the default configuration.
The two functions yield "Ok!" when I select Bluetooth Dual Mode:
```
CONFIG_BTDM_CONTROLLER_MODE_BTDM=y
```

devsaurus, 2019-05-31:
I don't see that we will change anything in bluetooth support soon.
Closing for now.

closed 2019-05-31 by devsaurus