nodemcu/nodemcu-firmware #2670: Can't boot ESP-12E in normal mode after flashing

alex3kov, 2019-02-15:
### Expected behavior
Normal operation.
### Actual behavior
ESP-12E doesn't boot after flashing.
### Test code
ESP8266 doesn't get to execution of Lua code. Here's what I did:
*  Connected GPIO0 to GND, connected to PC with CP2104 USB-UART connector, powered up ESP8266-12E with 3.3V (from separate power supply, not from CP2104)
*  Flashed NodeMCU (seems to be successful):
```sh
esptool --port /dev/ttyUSB0 write_flash -fm dio 0x00000 nodemcu-master-13-modules-2019-02-15-18-48-34-integer.bin 
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 18:fe:34:d6:5c:e1
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0240
Compressed 442368 bytes to 288369...
Wrote 442368 bytes (288369 compressed) at 0x00000000 in 25.4 seconds (effective 139.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
*  Disconnected ESP8266 from 3.3V, disconnected GPIO0 from GND, powered up ESP8266-12E with 3.3V (from separate power supply, not from CP2104). Tried resetting by connecting RST to GND - same result (below), here's boot log:
```
miniterm.py /dev/ttyUSB0 74880
--- Miniterm on /dev/ttyUSB0  74880,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 26424, room 16 
tail 8
chksum 0x23
load 0x3ffe8000, len 2196, room 0 
tail 4
chksum 0x2b
load 0x3ffe8894, len 136, room 4 
tail 4
chksum 0xe0
csum 0xe0
rf cal sector: 1019
freq trace enable 0
rf[112] : 00
rf[113] : 00
rf[114] : 01

SDK ver: 2.2.1(6ab97e9) compiled @ Jun  7 2018 13:59:14
phy ver: 1136_0, pp ver: 10.2
�␂�␂␐��	␆%���Q
!�␕���␑��P�Y��YJa�X�TP�\Y�\�Q�q�p␀�Y���
܇�ܧK␙␁%␆\␇	�ԇ�␑�	␕�␙␑␕E␂@�y
V␇IP␙h�␁QH�`K␁�I␝�␁!�R�I
```
* Tried to run esptool to get some basic info from chip (tried `-b 115200` as well):
```
esptool --trace --port /dev/ttyUSB0 -b 9600 read_mac
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.002 Read 1 bytes: ff
TRACE +0.000 Read invalid data: ff
TRACE +0.000 Remaining data in serial buffer: 
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
<plenty more "TRACE +0.000 Read invalid data: ff" blocks>
```
* Ran `esptool.py --port /dev/ttyUSB0 erase_flash` and repeated above steps - same result.
* Read the crap out of the docs, can't find what I'm doing wrong.
### NodeMCU version
Master, built on cloud 15.Feb.2019 with modules am2320, dht, ds18b20, file, gpio, mqtt, net, node, pwm, tmr, uart, wifi.

### Hardware
ESP8266-12E, no devices are connected to GPIO/ADC/etc pins. Just trying to flash it and get MAC address with `esptool read_mac`.

marcelstoer, 2019-02-16:
In general this is one for #1010.

> miniterm.py /dev/ttyUSB0 74880

Not sure if I missed it in your report but this should be 115200: https://nodemcu.readthedocs.io/en/latest/upload/

closed 2019-02-16 by marcelstoer

alex3kov, 2019-02-16:
> Not sure if I missed it in your report but this should be 115200: https://nodemcu.readthedocs.io/en/latest/upload/

According to `esptool` 74880 is for grabbing the boot log: https://github.com/espressif/esptool#boot-log

So, I should ask this on Stack?

TerryE, 2019-02-19:
Once the Boot ROM hands control to the NodeMCU runtime, the baudrate is switched by default to 115200. If you set `-b 74880` then the boot output will be captured but the NodeMCU output scrambled, and if `-b 115200` then v.v.

But this isn't a NodeMCU issue, so still closed.

alex3kov, 2019-02-19:
I had better luck with MicroPython, but thanks for the feedback anyway @marcelstoer @TerryE