nodemcu/nodemcu-firmware #2675: Make node.output() get called for interpreter errors

DexterLB, 2019-02-19:
### Missing feature
`node.output()` callback doesn't get called for interpreter errors.

### Justification
When debugging over WiFi (for example via telnet), it would be convenient to be easily able to see errors.

### Workarounds
External MOSFET which connects UART TX and RX, making it possible to read errors via `uart.on()`.

TerryE, 2019-02-22:
@DexterLB Angel, this has been discussed before.  It is impractical to do this for architectural reasons.  the `output()` function is supplied with a Lua function, but output to `stderr` is invoked at a low level in the Lua VM, and you will get all sorts of nasty re-entrancy issues if you try to invoke LuaVM at this level.  The correct approach here is to handle error handling _within_ your Lua application by using the `pcall()` or `xpcall()` baselib functions.

So example in my IoT apps, I have an 'execute lua command' and this code wraps the command in a pcall so that the application captures any errors and cleans up.

closed 2019-02-22 by TerryE

DexterLB, 2019-02-25:
Thanks! I hope this issue will be informative to future generations.