nodemcu/nodemcu-firmware #2681: Add msvc build to travis ci build

HHHartmann, 2019-02-24:
### Missing feature
Build hosttools for msvc

### Justification
not every developer will test it. if not tested regularily the build will decay over time

### Workarounds
none really

HHHartmann, 2019-06-24:
done since a while

closed 2019-06-24 by HHHartmann

TerryE, 2020-06-02:
@HHHartmann @nwf @marcelstoer, the subject of making a `luac.cross` available for windows users as a straight binary EXE file has been suggested on a number of occasions.  Simple to do, as far as I can see, but never done for some reason.  What is stopping this happening?  Should be open an issue to discuss this and the options? If so, should this be on this list or Marcel's cloud builder one?

marcelstoer, 2020-06-02:
If building it is easy (I don't know) then the only question is where to host the EXE. We can have Travis publish it to the wiki repo or to a dedicated branch in this repo.
I won't be building or hosting those files with the cloud builder.

TerryE, 2020-06-03:
> If building it is easy (I don't know)

@marcelstoer, that's what the MSVC build does: verify that we can make `luac.cross` with the MSVC toolchain. Surely just like your cloud builder script curl posts the bin files to https://nodemcu-build.com/archiver.php we could just as easily post the luac.cross exes to some service where we can make them available for download. If you don't want to use your website then perhaps we could use blog.ellisons.org.uk.

I just can't understand the unwillingness to make this available.

marcelstoer, 2020-06-03:
> we could just as easily post the luac.cross exes to some service where we can make them available for download.

Indeed. In addition to my earlier proposal "publish it to the wiki repo or to a dedicated branch in this repo" we could also create a dedicated repo to host the binaries here.

> I just can't understand the unwillingness to make this available.

I support making the binaries available, see above, just unwilling to host them with the cloud builder. The cloud builder creates personalized binaries; many are unique. The luac.cross EXEs, however, are tied to a specific master/dev branch version and only change when the code changes (potentially). If it's an official project offering then it should be hosted somewhere within our little GitHub universe as we don't have any other official online presence*.

\* not quite true if you consider nodemcu.com as per https://github.com/nodemcu/nodemcu-firmware/issues/2195#issuecomment-352260662