nodemcu/nodemcu-firmware #2683: http_request: If path is None, your delete \0 in URL

dontsovcmc, 2019-02-25:
So, we have function, that parse URL to components:
`void ICACHE_FLASH_ATTR http_request( const char * url, ....`

Ok, Google, please find '/':
```
char * path = os_strchr( url, '/' );
if ( path == NULL )
{
	path = os_strchr( url, '\0' );  /* Pointer to end of string. */
}
```  [624](https://github.com/nodemcu/nodemcu-firmware/blob/master/app/http/httpclient.c#L624)

Oookey, pointer to end of string.

Later:
```
if ( path[0] == '\0' ) /* Empty path is not allowed. */
{
	path = "/";
}
```
Come on, it's the end of the string!!

`HTTPCLIENT_DEBUG( "path=%s", path );` 

Wow, what about null terminal after lost null terminal?

nwf, 2019-02-25:
It's not immediately clear to me what you're asking, but you're coming across as incredibly rude and arrogant, even so.

I *think* you are confused and are acting as if line 662 (`path = "/";`) is, instead, `path[0] = '/';`, which it is emphatically not.  From a quick glance, as of line 664, path points either to 
* the suffix of the `url` beginning at the `'/'` separating the hostname from the path (and will therefore be just as `NUL`-terminated as `url` was) or
* to the literal string `"/"`, i.e. a `'/'` followed by a `'\0'`, i.e, something certainly `NUL`-terminated.

The dance with making `path` point to the `'\0'` within the `url` in case there isn't a `'/'` is so that subsequent comparisons and differences work out.  It's very classic C programmer.

If I've overlooked something, please advise.  If I *haven't* overlooked something, I expect an apology for the tone of this bug report.

dontsovcmc, 2019-02-25:
I'm sorry for my tone! I felt like found some important feature =/.
Yes, you are absolutely right about `path = "/";` is, instead, `path[0] = '/';`.. quotes.. 
Great, there is no bug!

closed 2019-02-25 by dontsovcmc