nodemcu/nodemcu-firmware #2690: Task watchdog got triggered.

maciejmiklas, 2019-03-13:
I am working on some Lua code for ESP32 https://github.com/maciejmiklas/NodeMCUUtils/tree/dev
Basically it feches json over WiFi and parsers it.

At the beginning I've used precompiled firmware, but I was unable to get UART1/2 working with it, so I've decided to compile latest dev firmware. The good news is, that UART now works fine, but I'm getting abort/reboot from watchdog when rading data from HTTP Server:

`[0;31mE (20076) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:[0m
[0;31mE (20076) task_wdt:  - IDLE0 (CPU 0)[0m
[0;31mE (20076) task_wdt: Tasks currently running:[0m
[0;31mE (20076) task_wdt: CPU 0: main[0m
[0;31mE (20076) task_wdt: CPU 1: IDLE1[0m
[0;31mE (20076) task_wdt: Aborting.[0m
abort() was called at PC 0x400d2a10 on core 0

Backtrace: 0x4008f037:0x3ffb09c0 0x4008f331:0x3ffb09e0 0x400d2a10:0x3ffb0a00 0x40082455:0x3ffb0a20 0x4013451a:0x00000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6032
ho 0 tail 12 room 4
load:0x40078000,len:8792
ho 0 tail 12 room 4
load:0x40080400,len:6244
entry 0x40080728
[0;32mI (32) boot: ESP-IDF v3.3-dev-149-g1c4d9f2ac 2nd stage bootloader[0m
[0;32mI (32) boot: compile time 09:49:06[0m
[0;32mI (41) boot: Enabling RNG early entropy source...[0m
[0;32mI (41) boot: SPI Speed      : 40MHz[0m
[0;32mI (43) boot: SPI Mode       : DIO[0m
[0;32mI (47) boot: SPI Flash Size : 4MB[0m
[0;32mI (51) boot: Partition Table:[0m
[0;32mI (54) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (77) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (84) boot:  3 spiffs           unknown          c2 00 00110000 002f0000[0m
[0;32mI (92) boot: End of partition table[0m
[0;32mI (96) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1f16c (127340) map[0m
[0;32mI (149) esp_image: segment 1: paddr=0x0002f194 vaddr=0x3ffb0000 size=0x00e7c (  3708) load[0m
[0;32mI (151) esp_image: segment 2: paddr=0x00030018 vaddr=0x400d0018 size=0x8b998 (571800) map[0m
[0;32mI (354) esp_image: segment 3: paddr=0x000bb9b8 vaddr=0x3ffb0e7c size=0x00e28 (  3624) load[0m
[0;32mI (355) esp_image: segment 4: paddr=0x000bc7e8 vaddr=0x3ffb1ca4 size=0x00000 (     0) load[0m
[0;32mI (360) esp_image: segment 5: paddr=0x000bc7f0 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (370) esp_image: segment 6: paddr=0x000bcbf8 vaddr=0x40080400 size=0x117a0 ( 71584) load[0m
[0;32mI (406) esp_image: segment 7: paddr=0x000ce3a0 vaddr=0x400c0000 size=0x00064 (   100) load[0m
[0;32mI (407) esp_image: segment 8: paddr=0x000ce40c vaddr=0x50000000 size=0x00000 (     0) load[0m
[0;32mI (413) esp_image: segment 9: paddr=0x000ce414 vaddr=0x50000000 size=0x00000 (     0) load[0m
[0;32mI (432) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (432) boot: Disabling RNG early entropy source...[0m
[0;32mI (434) cpu_start: Pro cpu up.[0m
[0;32mI (437) cpu_start: Starting app cpu, entry point is 0x400811cc[0m
[0;32mI (428) cpu_start: App cpu up.[0m
D (448) memory_layout: Checking 7 reserved memory ranges:[0m
D (453) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0[0m
D (459) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb7d90[0m
D (466) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440[0m
D (472) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350[0m
D (479) memory_layout: Reserved memory range 0x40070000 - 0x40078000[0m
D (485) memory_layout: Reserved memory range 0x40078000 - 0x40080000[0m
D (491) memory_layout: Reserved memory range 0x40080000 - 0x40091ba0[0m
D (498) memory_layout: Building list of available memory regions:[0m
D (504) memory_layout: Available memory region 0x3ffae6e0 - 0x3ffb0000[0m
D (511) memory_layout: Available memory region 0x3ffb7d90 - 0x3ffb8000[0m
D (517) memory_layout: Available memory region 0x3ffb8000 - 0x3ffc0000[0m
D (524) memory_layout: Available memory region 0x3ffc0000 - 0x3ffc2000[0m
[... 90 lines cut ...]
D (517) nvs: nvs_get bcn.interval 2[0m
D (527) nvs: nvs_get ap.phym 1[0m
D (527) nvs: nvs_get ap.phybw 1[0m
D (527) nvs: nvs_get ap.sndchan 1[0m
D (537) nvs: nvs_get lorate 1[0m
D (537) nvs: nvs_set_blob sta.mac 6[0m
D (557) nvs: nvs_set_blob ap.mac 6[0m
I (557) wifi: Init dynamic tx buffer num: 32
I (557) wifi: Init data frame dynamic rx buffer num: 32
I (557) wifi: Init management frame dynamic rx buffer num: 32
I (557) wifi: Init static rx buffer size: 1600
I (567) wifi: Init static rx buffer num: 10
I (567) wifi: Init dynamic rx buffer num: 32
NodeMCU ESP32 build unspecified powered by Lua 5.1.4
`

Here is decoded stack:

`Decoding stack results
0x4008f037: invoke_abort at /home/mmik/dev/opt/nodemcu-firmware-esp32/sdk/esp32-esp-idf/components/esp32/panic.c line 155
0x4008f331: abort at /home/mmik/dev/opt/nodemcu-firmware-esp32/sdk/esp32-esp-idf/components/esp32/panic.c line 170
0x400d2a10: task_wdt_isr at /home/mmik/dev/opt/nodemcu-firmware-esp32/sdk/esp32-esp-idf/components/esp32/task_wdt.c line 174
0x4013451a: reallymarkobject at /home/mmik/dev/opt/nodemcu-firmware-esp32/components/lua/lgc.c line 72`

I'am not sure what causes this problem, but maybe it would make sense to tweak default settings for make ?

devsaurus, 2019-03-15:
Your code doesn't work for me in ESPlorer. The line with `print(">> " ..` cannot be handled.
When I replace the `>>` with e.g. `:` everything works fine. There's no WDT panic for me.

```
boot: ESP-IDF v3.3-dev-149-g1c4d9f2ac 2nd stage bootloader
```
Why did you compile with ESP-IDF v3.3? The NodeMCU firmware currently uses commit 1c4d9f2ac on the v3.2 branch.

maciejmiklas, 2019-03-15:
I did not run it from explorer.  I've created `test.lua`, uploaded it and executed `dofile("test.lua")`. 

The code is from yesterday, I've been following: https://nodemcu.readthedocs.io/en/dev-esp32/build/ on ubuntu, so it's:
`git clone --branch dev-esp32 --recurse-submodules https://github.com/nodemcu/nodemcu-firmware.git nodemcu-firmware-esp32`

devsaurus, 2019-03-17:
> The code is from yesterday, I've been following: `git clone --branch dev-esp32 --recurse-submodules`

Then you get the same commit for esp-idf submodule?
```
$ cd sdk/esp32-esp-idf/
$ git clone --branch dev-esp32 --recurse-submodules
HEAD detached at 1c4d9f2ac
nothing to commit, working tree clean
```
> I've rewritten code to use `http` and still getting the same error. But sometimes it works, so you might need to try it few times.

I can't see anything obvious that would keep the nodemcu task busy for 5s so that the idle watchdog bites. Especially in case of the original implementation with `net` sockets.

We should rule out any inconsistency in your clone, thus if not already done, please try
1. `make clean`
2. Backup you `sdkconfig`, remove it and run `make defconfig`
    Check that `sdkconfig` got generated from scratch - e.g. diff against the backup.
    Don't change anything in this default config
3. Check correct submodule version in `sdk/esp32-esp-idf/`
4. `make flash` and try above code again

maciejmiklas, 2019-03-18:
this is mine:
```
mmik@uvm:~/dev/opt/nodemcu-firmware-esp32/sdk/esp32-esp-idf$ git status
HEAD detached at 1c4d9f2ac
nothing to commit, working tree clean
```
I've compiled the firmware as you've suggested (plus mqtt in order to upload files) and I cannot reproduce this error, but my version works also fine. Maybe network is faster or something. However I was able to modify my test case and I'm getting the same error again:

```
wifi.mode(wifi.STATION)
wifi.start()
wifi.sta.config({ ssid = '123', pwd = '123' })

local function test()
    con = net.createConnection(net.TCP)
    con:on("receive", function(cn, data)
        print(": " .. (string.len(data) / 1000) .. "kb, RAM: " .. (node.heap() / 1000) .. "kb")
        text = ""
        for i = 1, #data do
            local c = data:sub(i, i)
            text = c .. text
        end
    end)

    con:on("connection", function(sck, c)
        sck:send("GET /data/2.5/forecast?id=3081368&appid=3afb55b99aafbe3310545e4ced598754&units=metric" ..
                "  HTTP/1.1\r\nHost: api.openweathermap.org\r\nAccept: */*\r\n\r\n")
    end)
    con:connect(80, "api.openweathermap.org")
end

wifi.sta.on("got_ip", test)
```
It's definitely below 5 seconds

devsaurus, 2019-03-21:
My best guess at the moment is that there's an inconsistency within the firmware image.
Maybe your build results sometimes bump beyond the 1MB range which is provided by the default partition table.
A partition table with more space is located in `components/platform/partitions.csv`. You could try to select this as 'Custom partition table` in menuconfig.

maciejmiklas, 2019-03-21:
nope, this did not help.
But it seams that I am the only one with this problem, so there must be something wrong on my side. I will close this issue - thank you for help!

closed 2019-03-21 by maciejmiklas

maciejmiklas, 2019-03-27:
I've changed clock to 240MHz and the problem is solved.