nodemcu/nodemcu-firmware #2691: LFS Add storing arbitrary ressource files in LFS

HHHartmann, 2019-03-13:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and **DELETE THE REST**.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

### Actual behavior

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
-- add code here
```
### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.

### Hardware
Describe which ESP8266 device you use and document any special hardware setup 
required to reproduce the problem.

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

### Justification
Tell us why you would like to see this feature added.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

8<------------------------ END FEATURE REQUEST --------------------------------

galjonsfigur, 2019-03-13:
Sounds good to me! I was using LFS to store bitmaps before and it was working fine - I think this function would be very useful.

TerryE, 2019-03-13:
The workaround for now is that you can easily do this in Lua on your host PC now to generate a `resources.lua` that you can then compile into your LFS just the same way that you include `dummy_strings.lua`.  Just use either command line args or luaFS to loop around the list of resource files and use some naming convention for the resource name and then output the generated source to a `getresource.lua` file.  The trick is that the you can use the **`%q`** format specifier to do the heavy lifting.   
```Lua
source = source .. ('if arg = "%s" then return %q end'):format(name,resource)
```
On my TODO list is to add LuaFS and a server-side LFS library so you can also do this as part of the `luac.cross` using the `-e` option.

Then `LFS.getressource("someResourceName")` will return the Lua string containing the resource and this won't take any RAM as this resource will be in the `ROstrt` in the LFS image.

HHHartmann, 2019-03-14:
@TerryE terry I knew I had read an Idea about that from you somewhere.
I would prefer to have an array to access the ressources though. A long row of ```if``` statments is not optimal runtime wise.
Giving
```lua
LFS.ressources.someResourceName
```
or
```lua
LFS.ressources["some ugly name"]
```

I will change my implementation to use that interface, not polluting the LFS namespace.

> Note that you can only store strings this way but if you want more complex structures then just JSON encode them and return the decoded string

I do not see any reason why not to also have binaries here, Lua strings are 8 bit transparent aren't they?
Am I missing something?

I would also use the filename plus extension instead of just the base name. Thinking of a webserver (sorry) a "page.html", "page.js" and "page.css" might exist.
What are your thoughts about that.

TerryE, 2019-03-14:
> A long row of if statments is not optimal runtime wise.

If you have a module which contains 10 resources, then the runtime overhead is maybe 50 Lua VM instructions to access one.  Nothing really compared to opening and reading a SPIFFS file.  Using an if chain takes up no RAM, unlike declaring an array which does, even if the strings themselves are in LFS.

> I do not see any reason why not to also have binaries here

Yes, you can write binary data in a Lua string (even with embedded `\0` characters in it).  No my point was that of you want to keep an array resource, say, in LFS then you could always have the option of unpacking it into a returned temporary array.

TerryE, 2019-03-23:
@HHHartmann Gregor,  can I suggest that we close this issue in its current form?  I just don't see us implementing what you ask in its current form.  

In the short term we could add a host-side lua example to convert a list of arbitrary resources into a lua source for cross compilation into an LFS image.  The longer term might be to add Lua File System and luac library support to allow you to embed such conversion directly into `luac.cross` using the `-e` option.

HHHartmann, 2019-03-24:
@TerryE Terry I intended to implement it myself as outlined above. The suggestion I had above already works. I am currently working to adapt it to the ''''if then'''' implementation you suggested.

TerryE, 2019-03-24:
@HHHartmann Gregor, use the `-l` option on the cross compile to see what the code generator emits. I straight linear if chain will be the most efficient for resource lists up to about 8. Above that you might want to sort the resource array and do a binary chop, but probably not worthwhile as even using a long if chain will still be a lot more efficient than using SPIFFS.

TerryE, 2019-03-31:
@HHHartmann Gregor, sorry but I fundamentally disagree with the approach that you have adopted in the associated PR.  Given that we have already raised this conflict here, surely it would have been better resolving the issue here and agreeing an approach, so rather than ignoring this and carrying on regardless?  I will repeat my comment above: 

**Can I suggest that we close this issue in its current form? I just don't see us implementing what you ask in its current form.**

NodeMCU Lua currently implements a standard Lua 5.1 VM except for the ROTable extension introduced in eLua.  We have gone to some lengths to maintain this compatibility.  Your approach seems to extend and change the Lua VM in an adhoc and non-standard way.  Any resource can be embedded in Lua source code and therefore at runtime as a ROstst string, and the standard string library already provides the `%q` format specifier to facilitate any heavy lifting required to convert resources into standard Lua source representation.

Changes to the VM have all sort of knock-on ramifications, and need to be carefully thought through.  It makes no sense to extend / functionally modify the Lua VM when the VM already supports what we are attempting to implement.  If you wish to change or extend the VM then can I suggest that you first document your suggestions in a technical whitepaper, so we can agree a consensus approach _before_ attempting an implementation?   
 
Changing the Lua core is complicated and we should in any case only have one set of changes pending / in-progress.  I am currently holding off committing the SDK 3.0 PR until after the next master cut.  This change is going to add some 15Kb+ extra RAM to all Lua users.  Surely we should get this in and stable before we even start this work?

HHHartmann, 2019-03-31:
Sad to see your 👎 .

From reading

> Can I suggest that we close this issue in its current form? I just don't see us implementing what you ask in its current form.

I came to the conclusion that you guys don't want to spend time implementing it. So I made clear, that I was going to implement it.
You added comments and tips afterwards, so I read that as a Go.

I did,'t realize (my fault) that there was only a small changest to the original Lua implementation.
However I only touched luac.cross and the part in lauxlib.c which is ifdefed for luac.cross.

But anyway, I can still use it myself.

TerryE, 2019-03-31:
Gregor, you are approaching contributing to the NodeMCU core in the wrong way.  The project has to move forward based on consensus.

**Assertion**: You can achieve everything you want functionally without changing the Lua VM as this alredy facilitates everything that you want.

**Q**: Do you disagree with this statement and if so can you explain why?

**Q**:  If you accept this statement, then why on earth should we make architectural and implementation changes to the core? 

In this second case, especially as the final implementation is going to be more complicated and have more knock-ons.

If you are writing an add-on library for new devices or new functionality then we have more latitude, but when it comes to the core, we need ot debate and agree the functional requirements (FRs) and non-functional requirements (NFRs) _before_ we propose changes. 

Yes it might be easier for you to side-step reaching consensus and do your own fork, but it will remain that: your own fork and every time that we make other changes to the core, you are going to have to back-merge your patch or not use our updates.  Surely it would be better sorting out the FRs and NFRs first so we can agree an approach for the project?  and that way everyone can use your contribution.

If you do accept my assertion that you don't need to change the core, then I can see that we have three broadly separate approaches:
-  Use a small ~50 line lua script on the host to do the resource -> lua conversion
-  Add luac library support so that a lua script executed in `luac.cross` with the `-e` option can add modules to the generated LFS.  Note this functionality could be use to extend the compile time options in other ways as well.
-  Do as you suggest and statically extend `luac.cross` with some command line syntax such as your  `-r` option, but only modify luac and not the core.

HHHartmann, 2019-03-31:
Terry, glad to read from you.

I (now) see the problem in my approach. I somehow was not aware that the luac.cross was part of the lua core as it only gets executed on the host.

As to you assertion I can neither agree nor disagree due to lack of in depth knowledge. But reading your comments I tend to agree :-) 

Surely I would prefer to get this feature in the main repo and not have to maintain my own fork.

So we should get on sorting this out.

I think it to be nice to have only one step to perform and not have to run a separate script before generating the LFS image. Especially for unexperienced commandline users this would be another obstacle.

Your second option sounds tempting. This would result in something like
```luac.cross -f codefile.lua  -x GenerateRessourceModule.lua``` if I get it right. Just would need to be able to pass params to ``GenerateRessourceModule.lua``. This would be a major project for ma for which at the moment I would not find time to do.

I could move the changes out of lauxlib.c and into luac.c. Should I go for a PR about this?
So it would just be lexically moved out of the core. Functionally it is already.

TerryE, 2019-03-31:
The core Lua code for `luac` started life as a quick and dirty test hack, but proved so useful that it was hoisted into the core distro.  You can see this because the rest of the core code follows pretty strict coding style and structure conventions.  `luac` doesn't.  `luac.cross` was just an eLua hack of this.

The core `luac` only includes the compile and parse engine.  It doesn't include the execution engine.  I added the `-e` option and the execution engine initially for test purposes -- the LFS mods were pretty complex and very difficult to debug properly using the Xtensa remote debugger, so by adding the execution engine, I could debug the code using standard **gdb** with its far richer feature set on the host  and get that working before migrating to the ESP target.  However I was also future-proofing the ability to run a class of Lua scripts on the host without a separate lua install.

Before trying to implement anything we should agree the invariant requirements, for example:

-  **NFR1**: Don't change the execution of the core engine unless it is unavoidable and in which case minimise the change.  The is avoidable so I suggest that this should be: _This implementation should no require changes to the execution engine_.

Whilst there are reasons for not using a preprocessing Lua stage (in the finally deployed version) to convert the set of resource files into a single lua module, what would this look like if we were to do so?  It would be a simple Lua program:
```Lua
local a = ...
if a=='resourceName1' then return "<resource 1>" end
if a=='resourceName2' then return "<resource 2>" end
if a=='resourceName3' then return "<resource 3>" end
-- -- -- -- and so on
```
You can pass this through the `luac.cross -o /dev/null -l` command to see what code needs to be generated.  This is a module with 1 local, 2N string constants, and 2+4N instructions where the basic 4 instruction repeat is:
```
EQ       	0 0 -<2i>	; - "resourceName1"
JMP      	2	; to 6
LOADK    	1 -<2i+1>	; "<resource 1>"
RETURN   	1 2
```
where i is 0..N-1.  There are a couple of ways of generating this, such as just building a source version and then compiling it, or directly generating the code (see how I do this for the core dispatcher, and we could encapsulate and reuse this code.

As to the execution extension to the runtime, just implement this in Lua as part of `_init`.

Anyway, that's my first thoughts.  I've got to do a job here now, but will post back later.

HHHartmann, 2019-03-31:
What I am currently doing is generate just the Lua code you give above on the fly and feed it to the compiler. I just use ``if ... == "filename" then return "text" end`` instead of assigning a local variable first.
It generates just the code you mentioned above:

```
main <getresource:0,0> (11 instructions, 44 bytes at 00000206756E1300)
0+ params, 2 slots, 0 upvalues, 0 locals, 4 constants, 0 functions
        1       [1]     VARARG          0 2
        2       [1]     EQ              0 -1 0  ; "res1.txt" -
        3       [1]     JMP             2       ; to 6
        4       [1]     LOADK           0 -2    ; "Hello"
        5       [1]     RETURN          0 2
        6       [1]     VARARG          0 2
        7       [1]     EQ              0 -3 0  ; "res2.txt" -
        8       [1]     JMP             2       ; to 11
        9       [1]     LOADK           0 -4    ; "World"
        10      [1]     RETURN          0 2
        11      [1]     RETURN          0 1
```
Do I really have to do this in assembler like 
``` C
    *pc++ = CREATE_ABC(OP_EQ,0,0,RKASK(i)); 
```
style when the Lua compiler can do this for me?

> As to the execution extension to the runtime, just implement this in Lua as part of _init.

I don't see what is missing here. The generated code just has to be stored as ``getresource`` and then can be accessed like other modules too. And that's already in ``_init.lua``.
Or am am I missing something?

TerryE, 2019-03-31:
> Do I really have to do this in assembler ... ?

No, which is why I listed the option of calling the compiler.  I have to for the master dispatcher for reasons that I discuss in the LFS whitepaper [Implementation Details](https://nodemcu.readthedocs.io/en/master/en/lfs/#implementation-details) section.  Since this actually generates similar but not the same VM code, it is probably not worth encapsulating this generator code in a function to call it a second time, but having a `generate_resources()` based on `combine()` might be the cleanest way.  Just collect a Lua table of name = "string value" and pass it as an argument. 

I'll email you a rough cut tomorrow for you to look at.  

Partly my whole approach to modifying the core is to brood about alternative implementation options and then pick the one that involves the least impact implementation.

TerryE, 2019-04-01:
@HHHartmann Gregor, let us assume for the moment that we can implement this functionality with some changes in `luac.c` and the LFS `_init` modules, I come back to my points on this issue and that is the FRs for this.

Lua as a language can only embed numeric, string, and function prototype constants in compiled code. So unlike PHP (for example) where you can embed constant arrays in the compiled code, Lua _must_  build these at runtime as well as binding function prototypes at closure to create function variables.  We change this at our peril, IMO.  So what this issue is really about is whether there is a broad consensus amongst the committers and active contributors that 
-  we should extend `luac.cross` to allow it to generate a NodeMCU-specific source module which embeds a resource function such that if you call it with a resource name then it will return a (ROstrt) string containing the corresponding content of the file with that name.

As I mention above: we have four broad approaches options:
1.  Don't bother and avoid a lot of change. 
2.  Provide a Lua (or Python) example script than can be run standalone on the host PC to do this set of resource files -> Lua source conversion.
3.  Hard-code the equivalent translation function into `luac.cross`
4.  Add sufficient library support to luac.cross (os, io, luafs, lfs) so the we don't need to hardcode the translation, but so can execute a Lua script equivalent to (1) within `luac.cross`  using the `-e` option.

I personally think that procs and cons for this change tip on the side of the cons.  Being able to still resources into LFS as RO string constants might seem like a good idea, but quite honestly using SPIFFS to store them is just as simple, and there isn't any material performance advantage in moving these resources from SPIFFS to LFS.

If we do this then I can imagine novice developers getting into all sorts of issues as doing this will burn LFS region use.  The more advanced users are quite capably of running a Lua or Python pre-processor to do this conversion and we could even add a make target to encapsulate it.

I note that we haven't even bothered to implement a simple `file.getcontents(name)` which is a **lot** simpler to implement and would be a lot more widely used.

If we do chose the option of a NodeMCU extension to  _extending `luca.cross`_ then there are details that we should agree on such as the command line syntax and ensuring that we avoid backwards compatibility issues.  I think that the syntax being proposed is:
```
luac.cross [options except -r] [Lua filenames] [-r [resource filenames]]
```
Is the best way of doing this?

My feeling is that we should support any active committer is extending the firmware library with non-core changes so long as they are implemented to acceptable standard.  I also feel that we need to be far more cautious about extending the Lua core and runtime system.  There should be clear and broadly applicable advantages over the workaround alternatives.  We need a majority of the active committers supporting the need for this change.

TerryE, 2019-04-05:
Gregor, in our offline discussions, I think that we've identified a safe way ahead for this. I will leave you to explain and to raise a new PR to cover this work.

closed 2020-08-29 by stale[bot]