nodemcu/nodemcu-firmware #2698: dev-esp32 branch: cross-lua.lua is broken by recent changes

crdev, 2019-03-18:
Execute `$ lua tools/cross-lua.lua`

### Expected behavior
Build successful

### Actual behavior
```
[COMPILE] .build/cross-lua/app__lua__lapi.o
gcc: error: app/lua/lapi.c: No such file or directory
gcc: fatal error: no input files
compilation terminated.
[builder] Error building target
[builder] Last executed command was: 
gcc -O2 -Iapp/include -Iinclude -Iapp/lua -Wall -DLUA_CROSS_COMPILER -c -MD -MF .build/cross-lua/app__lua__lapi.d app/lua/lapi.c -o .build/cross-lua/app__lua__lapi.o
```

### NodeMCU version
branch dev-esp32, commit d1eab2396cc4c012b1f5e29fb6a79bd90637f97d

### Hardware
ESP32 Mini

devsaurus, 2019-03-21:
> `gcc: error: app/lua/lapi.c: No such file or directory`

I guess this script was never adapted to the esp32 branch. Include reside in `component/lua`, not `app/lua`.

Does it help anything when you fix these paths?

devsaurus, 2019-05-31:
@crdev any updates?

jmattsson, 2019-06-01:
One of my colleagues is currently working on porting the LFS support over to the ESP32. That will bring in the new luac.cross compiler too. Stay tuned, hopefully there'll be a PR this coming week.

crdev, 2019-06-02:
Sorry, I've moved off the project but IIRC the last time I tried replacing the directory name I kept getting some link-time errors.