nodemcu/nodemcu-firmware #2701: Unaligned load/store exception in luaR_findentry

pjsg, 2019-03-23:
### Expected behavior

it doesn't trap!

### Actual behavior

It traps

### Test code
With LFS enabled and a flash image loaded with @TerryE _init.lua loaded and initialized and with the string "/data" in the LFS

```Lua
require (string.sub("/data", 2, -1))
```
### NodeMCU version
Current dev

I'm not sure which instruction blows up.

TerryE, 2019-03-23:
@pjsg Philip, I'll take a look, but I might need the code base for a test case.  I will see if I can recreate it just from the info posted.

pjsg, 2019-03-23:
I just diagnosed the issue -- I updated the original message -- it is a lack of a l32ui unaligned handler.

TerryE, 2019-03-23:
That's because we shouldn't be doing any unaligned 32-bit accesses. If we are then it's a bug, because all of our 32 and 64-bit fields would be word-aligned.

pjsg, 2019-03-23:
The luaR_findentry code assumes that strings are 32 bit aligned ...

TerryE, 2019-03-23:
I know that. Are you compiling with `-Os`?  That would explain the error, but even in this case, we should still tweak the macros to force this as `-Os` should only result in a performance hit and not an error.

pjsg, 2019-03-23:
No. -O2. The issue is (I think) that the string.sub can return an unaligned
string if you are requesting the tail end of a read-only string.

TerryE, 2019-03-23:
Thanks Philip, I'll look, and I think that you might be close to the nub. Standard Lua creates a new TString with an aligned copy of the string. The issue is that we are using the eLua patch which uses a reference if the original string is in Flash. I'll push out a fix tomorrow.

TerryE, 2019-03-24:
OK, `str_sub()` does a `lua_pushlstring(L, s+start-1, end-start+1)` in other words if the string is in ROM then, this passes the (not necessarily word aligned) start address and length.  This is just a parameter checking wrapper for a call to `luaS_newlstr(L, s, len)` which calls `newlstr(L, str, l, h, TRUE)` if this is a new TString where the string is in Flash.  With the **eLua** patch, the TString uses a reference to the (not necessarily word aligned) start address and sets the `marked` READONLY bit to flag this use of a reference.

The issue is that the ROTable scan starting at [lrotable.c:L104](/nodemcu/nodemcu-firmware/blob/dev/app/lua/lrotable.c#L104) has been highly optimised to do one flash word read per table entry and the assignment at L104 assumes that strkey is word aligned.  Oops.  The simplest work around that keeps this fast scan is to use `c_memcpy()` for this assignment as this ROM version is optimised for word aligned copies but still handles non-aligned copies from ROM correctly.

TerryE, 2019-03-24:
`_VERSION` is another string using a RO indirection so `require ('_VERSION'):sub(3,-1)` also does the required crash.  The `c_memcpy` change fixes this.  I was sloppy and just used an SDK 3.0 version first to test this bug, but just couldn't get it to raise the exception.  Guess what the SDK 3.0 uses an Espressif version of the exception handler and this _does_ handle unaligned 32bit errors.  No crash here.  Still, it is worth putting in this one liner.

PR #2702 (2019-03-24): Fix unaligned load/store exception in luaR_findentry
Fixes #2701

- [X] This PR is for the `dev` branch rather than for `master`.
- [X] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [X] I have thoroughly tested my contribution.
- [X] No documentation changes required at `docs/en/*`.

Removes the possible unaligned fetch in `luaR_findentry()`

@marcelstoer, IMO we should add this to the current dev->master cut.
closed 2019-03-25 by TerryE