nodemcu/nodemcu-firmware #2703: crypto.mask on a zero length string says 'e:m 0'

pjsg, 2019-03-24:
### Expected behavior
It should just return an empty string
### Actual behavior
It prints out 'E:M 0'
### Test code

```Lua
> =crypto.mask("", 'bar')
E:M 0

```
### NodeMCU version
dev

This is because the code looks like:

```c
  int len, mask_len;
  const char* msg = luaL_checklstring(L, 1, &len);
  const char* mask = luaL_checklstring(L, 2, &mask_len);

  if(mask_len <= 0)
    return luaL_error(L, "invalid argument: mask");

  int i;
  char* copy = (char*)c_malloc(len);

  for (i = 0; i < len; i++) {
    copy[i] = msg[i] ^ mask[i % mask_len];
  }
  lua_pushlstring(L, copy, len);
  c_free(copy);

  return 1;

```
and calling c_malloc(0) causes this to be printed....

TerryE, 2019-04-05:
Actually using `malloc()` is the wrong way to do this in Lua C modules for a couple of reasons:
-  first there is no allocation fail checks
-  using a Lua buffer is the right way to do this. I will document why in a separate issue but the tl;dr is that it is in general more memory efficient and does proper Lua error handling and GC.

I will do the PR.

TerryE, 2019-04-05:
@jmattsson Johny I decided to remove the potential malloc leaks in this code (no return status error paths) by just using `luaL_Buffer` templates so that Lua handles all of this crap and correctly handles things like empty strings.  

The only only other allocator is for the USER data because you've also got a context structure and a HMAC opad to allocate as well.  You then need a **__GC** metamethod to handle freeing these.  Since these only get freed at GC, isn't just easier to make the UD be a composite of all three with the size just the sum of the sizes.  This unifies allocation in the single `lua_newuserdata()` call and so we don't need malloc failure error handling.  Ditto we don't need the `__GC` method since the standard GC will now free the composite block.

TerryE, 2019-04-05:
I went through all of test cases in the docs, and picked up a few typos in the examples.  Also the one above.  

It feels odd having `node.heap()` return `59592` :smiley:

PR #2712 (2019-04-08): Crypto fix EM:0 error, and other memory leaks, plus minor typos in doc
Fixes #2703

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

Bugfix and tidy up as discussed in #2703
closed 2019-04-09 by marcelstoer