nodemcu/nodemcu-firmware #2711: DOCS: Should bme280 example recommend different pins?

FGasper, 2019-04-04:
### Expected behavior

The example code should yield normal operation.

### Actual behavior

Using pins 3 & 4 for SDA and SCL as given in the bme280 example caused the LED on my ESP8266 to blink every time I ran `bme280.baro()`.

### NodeMCU version
```
NodeMCU custom build by frightanic.com
	branch: master
	commit: 11592951b90707cdcb6d751876170bf4da82850d
	SSL: false
	modules: bme280,file,gpio,i2c,mqtt,net,node,rtctime,sntp,spi,tmr,uart,wifi
 build created on 2019-02-27 04:49
 powered by Lua 5.1.4 on SDK 2.2.1(6ab97e9)
```

### Hardware
nodemcu dev board w/ 4 MiB of flash memory.

vsky279, 2019-04-12:
Hello Felipe,
Do you bme280 connected to GPIO0 (IO index 3) and GPIO2 (IO index 4)?
Please see https://nodemcu.readthedocs.io/en/dev/modules/gpio/ for GPIO mapping.

FGasper, 2019-04-12:
@vsky279 Thanks for responding!

I understand the GPIO mapping and why the blink happens. But it took me quite a while to realize what was going on, and it required that I do more research than was needed just to use the bme280 with nodemcu.

I had bme280 connected to D3 and D4 on the board, which allowed the demo code to work; however, that also produced the undesirable blink of the flash LED every time the bme280 was queried.

My proposal is just that the demo should say `sda, scl = 1, 2` rather than `sda, scl = 3, 4` because using D3 & D4 causes the blink, which is excess, likely undesirable behavior.

Or, at least there should be a comment that says, “if you don’t like the blink, use different IO pins than D3 & D4.”

Does that sound reasonable?

vsky279, 2019-04-12:
@FGasper It seems to me that your issue is specific to the hw you are using, i.e. nodemcu dev board.
I don't have led blinking with ESP8266 ESP-01 nor with ESP8266 ESP-12E.

I think that you would have the blinking issue with any other module accessing GPIO0 or GPIO2 (not sure which one is the one connected to the led). So it is not specific to BME280 module. I don't think it is appropriate to comment this behavior in the module documention (not even in I2C module documentation). 

`sda, scl = 3, 4` are used deliberately as these the only GPIOs available on ESP-01 which I think used to be the most common variant of ESP8266 (and I'm still using it).

I am glad that you found the reasons for the undesired blinking anyway. :-)

I propose closing this issue.

FGasper, 2019-04-12:
Ahh, ok.

I would still think it useful to comment since the nodemcu dev board is very popular, but of course it’s your call.

Anyhow, thank you for maintaining this project! It’s such fun to use.

-F

vsky279, 2019-04-13:
Let's wait for an opinion of core developers.

Your thanks go to these guys as I contributed only few lines of code :-)

marcelstoer, 2019-04-13:
> I think that you would have the blinking issue with any other module accessing GPIO0 or GPIO2 (not sure which one is the one connected to the led). So it is not specific to BME280 module. I don't [...]

👍If we were to point out the differences between GPIO pin number and index here (which we do in the GPIO documentation) we would have to do it for _every single example_.

closed 2019-04-13 by marcelstoer