nodemcu/nodemcu-firmware #2714: esp8266 wdt reset during smartconfig

psatya111, 2019-04-07:
### Expected behavior :- should not get the hardware watch dog reset during smart config.

### Actual behavior :- During smart config esp8266 is getting stuck in smartconfig  SDK code and hardware watch dog is getting triggered.

### Test log:

All API call after smartconfig in mainloop *************
All API call after smartconfig in mainloop *************
8:35:50
ReserCountTimeOut:
ResetCount: 0
ResetWifiMode: 6
Reset Done wifi configured to STATION MODE. SC version: V2.5.4
Smartconfig Started...
Esp8266WIFI::ResetDefaultSTAMode() called
WIFI MODE RESET TO STA
state: 5 -> 0 (0)
rm 0
pm close 7
wifi evt: 8
wifi evt: 1
STA disconnect: 8
scandone
scandone
ReserCountTimeOut:
ResetCount: 0
ResetWifiMode: 0
ReserCountTimeOut:
ResetCount: 0
ResetWifiMode: 0
ReserCountTimeOut:
ResetCount: 0
ResetWifiMode: 0
ReserCountTimeOut:
ResetCount: 0
ResetWifiMode: 0
ReserCountTimeOut:
ResetCount: 0
ResetWifiMode: 0
ReserCountTimeOut:
ResetCount: 0
ResetWifiMode: 0
ReserCountTimeOut:
ResetCount: 0
ResetWifiMode: 0
ReserCountTimeOut:
ResetCount: 0
ResetWifiMode: 0
ReserCountTimeOut:
ResetCount: 0
ResetWifiMode: 0
ReserCountTimeOut:
ResetCount: 0
ResetWifiMode: 0
ReserCountTimeOut:
ResetCount: 0
ResetWifiMode: 0
ReserCountTimeOut:
ResetCount: 0
ResetWifiMode: 0

TYPE: ESPTOUCH

T|sniffer on ch:4
T|PHONE MAC: 08 25 25 36 a3 b9
T|AP MAC   : 30 b5 c2 2e 9f 2c

T|Head Len : 82

T|SYNC STATUS

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld

SDK:3.0.0-dev(c0f7b44)/Core:2.5.0=20500000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1/BearSSL:6778687
after delay of 2s
/tmp/arduino_build_245637/sketch/Main.cpp20
ESP CORE VERSION :2_5_0
ESP SDK VERSION :3.0.0-dev(c0f7b44)
-1
Satyabrata

### version:-
SDK:3.0.0-dev(c0f7b44)/
Core:2.5.0=20500000/
lwIP:STABLE-2_1_2_RELEASE/
glue:1.1/BearSSL:6778687
 SC version: V2.5.4

Please help me .

EgorOA, 2019-04-14:
Hi!
I have wdt reset during node.compile() of lua file (13.5kB) on the newest (14/04/2019) firmware

TerryE, 2019-04-24:
@psatya111, your issue has nothing to do with a nodemcu-firmware build, so this is the wrong issues list.
@EgorOA please open your own issue complying to the template.

closed 2019-04-24 by TerryE