nodemcu/nodemcu-firmware #2716: wifi.suspend() suspend_cb function not called when suspending, resume_cb called when resuming

poorandunlucky, 2019-04-08:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

I was expecting the suspension callback function in `wifi.suspend()` to run at the same time as `wifi.suspend()`, but the function I registered there didn't run.

### Actual behavior

The `resume_cb` function is called, and seems to be fine, but nothing was printed when I used `wifi.suspend()`, at least on the first call...  I don't know about subsequent calls.  as `suspend_cb` is part of an optional series of arguments.

### Test code

`wifi.suspend ( 0, resume_cb = function ( ) print ( 'resumed' ), suspend_cb = function ( ) print ( 'suspended' ), false )`

### NodeMCU version

2.2.0 Master

### Hardware

ESP8266MOD

8<------------------------ END BUG REPORT -------------------------------------

poorandunlucky, 2019-04-09:
Actually there was a problem in my callback function...  : <

I just hooked a USB-UART thing to the RX/TX pins of the MCU and I got the error while calling the function, `wifi.suspend()` was returning -1...

Well done!  ^_^'

TerryE, 2019-04-09:
There's a problem with your Lua syntax for a start.  I would expect something like:
```Lua
wifi.suspend ( 0, 
    function ( ) print ( 'resumed' ) end, 
    function ( ) print ( 'suspended' ) end, 
    false )`
```
An assignment staatement is not allowed to be embedded in an arglist in Lua.

poorandunlucky, 2019-04-09:
@TerryE I'm sorry for the confusion...  I was working on too many things at the same time...

There is an actual problem with the wifi.suspend() function where it won't accept the call I was trying to pass, my guess for being too long...  wifi.suspend() actually takes a table as an argument, but my example code was missing brackets...

I edited my original post so everything is clear, and exact...  There shouldn't be any errors now.

TerryE, 2019-04-10:
Yes, you are correct, suspend takes a hash, but it should only contain keyed entries:
```Lua
wifi.suspend ({
    duration = 0, 
    suspend_cb =  function ( ) print ( 'resumed' ) end, 
    resume_cb = function ( ) print ( 'suspended' ) end, 
    preserve_mode = false })
```
so for example omitting the `duration =` results in settiing the entry [1] to zero so the routine will report a missing duration, etc.

I am closing this for now.  Please reopen if needed -- once you've eliminated your own bug and are convinced that you are left with an underlying issue with the wifi module.

closed 2019-04-10 by TerryE