nodemcu/nodemcu-firmware #2720: Error from `tools/Makefile`

devsaurus, 2019-04-11:
### Expected behavior
Compile from `dev` branch passes.

### Actual behavior
```
ESPTOOL app/.output/eagle/debug/image/eagle.app.v6.out ../bin/
esptool.py v2.6
Creating image for ESP8266...
make[1]: Leaving directory 'nodemcu-firmware_devsaurus/app'
make[1]: Entering directory 'nodemcu-firmware_devsaurus/tools'
/bin/sh: -c: line 0: syntax error near unexpected token `('
/bin/sh: -c: line 0: `printf "0x%x" (~0x0)'
```

### Test code
```
make
```

### NodeMCU version
`dev` at 5a6992c26a45a730edb2fff828e1edba16ef858d

### Additional information
The error is triggered by https://github.com/nodemcu/nodemcu-firmware/blob/5a6992c26a45a730edb2fff828e1edba16ef858d/tools/Makefile#L26

The value for `$(FLASH_FS_SIZE)` is not accepted: `(~0x0)`

```
../tools/toolchains/esp8266-linux-x86_64-20181106.0/bin/xtensa-lx106-elf-gcc -E -dM - <../app/include/user_config.h | grep SPIFFS_MAX_FILESYSTEM_SIZE
#define SPIFFS_MAX_FILESYSTEM_SIZE (~0x0)
```

TerryE, 2019-04-11:
Thanks Arnim, we need to rethink this logic anyway for SDK 3.0.  I'll pick it up on my next set of patches for the new SDK.

TerryE, 2019-05-06:
@devsaurus, I've replaced the default value of `~0x0` by `0xFFFFFFFF` which removes this error, so that ``FLASHSIZE` gets set to a valid value.  We still need to put the correct logic in here though.

TerryE, 2019-10-28:
Having done the previous change, the residual issue is now covered in #2943 so let's just keep the one tracking issue open.

closed 2019-10-28 by TerryE