nodemcu/nodemcu-firmware #2727: luac.cross -f fails if an lc file is included on the command line

TerryE, 2019-04-15:
@cwrseck posted in re #2656 comment
> Looking at the LFS build process, eg: luac.cross -f -o lfs.img *lua, it's not clear whether or not the .lua files are simultaneously compiled to .lc. The LFS._index method just gives the module name, not the extension. It's probably worth adding a comment to the docs clarifying the actual behaviour.

the answer is no; luac.cross of 20190405 segfaults when given an `.lc` file to include in an lfs image.  I don't know if this is a bug or a feature.

C W Rose

_Originally posted by @cwrseck in https://github.com/nodemcu/nodemcu-firmware/issues/2656#issuecomment-483058111_

TerryE, 2019-04-15:
@cwrseck Will, it would help us to respond constructively to your comments if you could follow the guidelines detailed in #1010.  Specifically: this should be a separate issue; please use MD; and provide a repeatable test case, for example:
```Bash
> cd $NODEMCU-FIRMWARE
> ./luac.cross -o /tmp/_init.lc lua_examples/lfs/_init.lua
> ./luac.cross -o /tmp/lfs.img  lua_examples/lfs/dummy_strings.lua /tmp/_init.lc
Segmentation fault (core dumped)
```
Thanks.

I can't think of any reason why you would want to include a pre-compiles Lua file rather then just Lua sources, but off the top of my head, I can't see why this should fail.  The workaround is simply to use the source version as input.  However I will add this to my TODO list and gather this up next time I do some updates to luac.cross.  If you want to track it, then you will need to raise this as its own issue.

TerryE, 2019-04-15:
@TerryE if it helps at all, I did reproduce the crash when doing a luac.cross with a pre-compiled Lua chunk.  FWIW, I found the crash to fundamentally be at:
`lauxlib.c:738`
in the `getF()` implementation.  This is because the `size` parameter is NULL, and is being deref'ed.
Going up the stack, this is in turn because at:
`lundump.c:238 `
in the `LoadFunction()` implementation there is code ` if (luaZ_direct_mode(S->Z)) l_setbit((f)->marked, READONLYBIT);`.  The `luaZ_direct_mode` is a macro that in turn uses another macro `luaZ_get_base_address` which performs `(zio)->reader()`.  `reader` in this case is pointing to the getF() function, and it explicitly passes NULL for the first and last parameter (the first parameter is unused in getF(), but the last one is the 'size' param mentioned before, hence the crash).

I also built the stock Lua 5.1.4 code, and observed this crashing did _not_ occur.  Looking at that same LoadFunction() code, I notice that the line involving `luaZ_direct_mode` is _not_ present, so I assume it is something that was added for eLua.

I'm not familiar enough with the eLua codebase to make any suggestions for a fix, but I thought this diagnostic info might be useful for someone who is more familiar.

I do agree that a separate bug filing should be made for this, but I wanted to at least record this info here so it could be transferred to such filing if this info is considered useful.

_Originally posted by @ziggurat29 in https://github.com/nodemcu/nodemcu-firmware/issues/2656#issuecomment-483085911_

TerryE, 2019-04-15:
@ziggurat29, this one got me intreged as well and I did the same exercise as you and came to the same conclusion, though I didn't write it up because this issue is a documentation one. I suspect that the underlying issue is a subtle cross compile one causing the undump to fail.

ziggurat29, 2019-04-15:
@TerryE I did a little spelunking with 'blame' on both nodemcu and eLua in the lauxlib.c around the getF() implementation.  Some things of note:
*  eLua has special code handling the NULL, NULL parameter case, commented `// special request: detect 'direct mode'`, and special handling there and later.
*  nodemcu does not have this special case handling (and that is the source of the crash).

I notice the commit in eLua that adds this code was made in 2012, which to wit was before nodemcu came about (2014), so it should have been in the codebase at the time of derivation so I suspect it was probably removed conscientiously (albeit probably also in error in some way).

As best as I can tell, the eLua additions were to support 'ROMFS' (is this some sort of XIP capability?), which sounds conceptually similar to LFS (I could easily be wrong), so maybe some of that logic could be reused.  The logic seems to hinge upon some extra fields added to the (private) `struct LoadF` that refer to the file not as a file, but as some sort of direct access buffer.  I don't know where this buffer is, it doesn't seem to be allocated here, so maybe it is pointing into something allocated elsewhere that is ROMFS related (it comes from `dm_getaddr()` which is not in the nodemcu codebase).

Anyway, that's as much as I've figured out for now.  And one alternative is to simply refuse to accept *.lc files, either way back in main during parameter validation, or maybe in ldo.c f_parser() where the lc file format is detected and switched to direct luaU_undump()'ing rather than luaY_parser()'ing  (I don't know how you would communicate an error there, though).

TerryE, 2019-04-15:
@galjonsfigur thanks for this.  ROMFS is sort of a halfway house between SPIFFS and LFS.  With ROMFS code, all of the meta structures like the Proto records and the TString records are still stored in RAM, but the actual Cstr and code vectors are accessed from addressable ROM.  LFS keeps all of the TString and Proto records in addressable ROM and has no code and string constant related RAM overhead; more of a true modified Harvard architecture.  We don't use any of this ROMFS complication, but we left it in because it was just easier to make the minimum changes to the eLua patch.

cwrseck, 2019-04-16:
@TerryE - I didn't raise a bug report because, as I said in my message,
I didn't know if the behaviour was a bug or a feature.  On the whole,
I decided it was a feature, but one worth reporting in the documentation.

I wanted to add .lc file to an LFS image to minimise memory usage
on the ESP8266, though with the current 40K+ memory space that's not
really necessary any more.

C W Rose

TerryE, 2019-04-16:
> I wanted to add .lc file to an LFS image to minimise memory usage on the ESP8266,
@cwrseck, Will, 

I am not sure what you are meaning by this especially as to seem to be attempting to add `lc` files.  The purpose of **luac.cross** is to convert one or more sources into a **_compiled_** format.  

In the case of an LFS image generated with the `-f` option,  you can't store uncompiled sources, only compiled ones. (See the `-l` option to get an assembler listing if you want).  Once this image is loaded into the LFS region, this set of Lua modules / files all run directly out of flash. 

If you want a better understanding of the internals, read the LFS whitepaper which how LFS works; also the FAQ also links to **A No Frills Introduction to Lua 5.1 VM Instructions** [here](https://nodemcu.readthedocs.io/en/dev/lua-developer-faq/#how-do-i-get-a-feel-for-how-much-memory-my-functions-use). This explains how the code generator works, how much memory overhead is involved with each table, function, string etc.  

With the 3.0 version, you can have up to 256Kb of compiled Lua sitting in your LFS region and this still leaves ~59Kb RAM for variables / data.

cwrseck, 2019-04-17:
@TerryE Excellent, thanks, that makes everything much clearer.

C W Rose

TerryE, 2019-10-28:
Fixed in #2912

closed 2019-10-28 by TerryE