nodemcu/nodemcu-firmware #2728: About manually compiling NodeMCU 2.2.1 firmware added u8g2lib there is no problem with u8x8.h this file

chenlugithub, 2019-04-15:
About manually compiling NodeMCU 2.2.1 firmware added u8g2lib there is no problem with u8x8.h this file.

8<------------------------ BUG REPORT -----------------------------------------
DEPEND: CC app/u8g2lib/u8x8_d_fbrle.c
u8x8_d_fbrle.c:4:18: fatal error: u8x8.h: No such file or directory
 #include "u8x8.h"
                  ^
compilation terminated.
CC app/u8g2lib/u8x8_d_fbrle.c
u8x8_d_fbrle.c:4:18: fatal error: u8x8.h: No such file or directory
 #include "u8x8.h"
                  ^
compilation terminated.
../../Makefile:366: recipe for target '.output/eagle/debug/obj/u8x8_d_fbrle.o' failed
make[2]: *** [.output/eagle/debug/obj/u8x8_d_fbrle.o] Error 1
make[2]: Leaving directory '/root/esp/nodemcu-firmware-master/app/u8g2lib'
../Makefile:327: recipe for target '.subdirs' failed
make[1]: *** [.subdirs] Error 2
make[1]: Leaving directory '/root/esp/nodemcu-firmware-master/app'
Makefile:327: recipe for target '.subdirs' failed
make: *** [.subdirs] Error 2
make: Leaving directory '/root/esp/nodemcu-firmware-master'
root@cl-virtual-machine:~/esp/nodemcu-firmware-master# About manually compiling NodeMCU 2.2.1 firmware added u8g2lib there is no problem with u8x8.h this file

About: command not found

8<------------------------ END BUG REPORT -------------------------------------

devsaurus, 2019-04-16:
You downloaded the ZIP file and also have to manually download the submodules like u8g2lib.

Only a git clone will get you a complete, consistent source tree:
https://nodemcu.readthedocs.io/en/master/build/#git

marcelstoer, 2019-04-17:
See #1010

closed 2019-04-17 by marcelstoer

chenlugithub, 2019-05-08:
> You downloaded the ZIP file and also have to manually download the submodules like u8g2lib.
> 
> Only a git clone will get you a complete, consistent source tree:
> https://nodemcu.readthedocs.io/en/master/build/#git

i dont undantand you mean. i add u8g2 models in app/include/user_model.h.  but it not succes i am sad

TerryE, 2019-05-08:
> i dont undantand you mean.

The graphics libraries are separate Git projects. We used to keep copies of this code in our project, but this is not the recommended Git practice.  If you use these libraries and do a ZIP download to get the firmware, then you also need to do the same from the graphics library repository.

>