nodemcu/nodemcu-firmware #2729: Build against recent SDK version rather than original release

marcelstoer, 2019-04-17:
Follow-up to https://github.com/nodemcu/nodemcu-firmware/issues/2689#issuecomment-484037820

We used to pull in the latest revision of an SDK IIRC. With 3.0 we seem to have switched to using the initially released version.

Unless there are really good reasons that speak against it we should be using the latest revision to profit from any upstream fixes.

> so that the builder can decide whether to select the Release SDK or the latest (as at the Makefile)

I disagree. We should stick to a version we are happy with. That way we can provide a known to be working combination to our users. I don't think it help contribute to our effectiveness if we had to handle issues here that really are SDK issues. Whatever breaks will be reported here.

TerryE, 2019-04-17:
Marcel, I have a slightly different perspective.  I know from experience that I get little testing feedback from commits to my fork, and some of these changes are just so complex that we do need to exercise them and this ends up making `dev` a true development environment for big changes like SDK 3.0.  If we want testers to form a view as to whether the formal Release or the latest commit to the 3.0 branch are better / more stable then I think that we just won;t get this feedback unless building the two variants is painless.  We are talking about an extra ½ dozen lines of Makefile to do this with an optional argument to select the non-default option.

marcelstoer, 2019-04-18:
I understand Terry. I now see that my comments aren't clear enough.

- I'm ok to make it configurable if you feel that it would help us
- I propose to change the default, my main issue here, to the _latest_ (at the time of inclusion of course)

TerryE, 2019-05-13:
As an update here: I have a working copy of the `app/Makefile` which introduces an extra make option `RELEASE` with the current options:
-  **(omitted)**. This is the current default and is pulled the official tagged [v3.0](/espressif/ESP8266_NONOS_SDK/releases/tag/v3.0) release. _However_, IMO this manifests a know show-stopper (see #2751 and espressif/ESP8266_NONOS_SDK#243) which causes a race condition and WDT on memory exhaustion, so **_we shouldn't ship based on this SDK build_**.
-  **`RELEASE=master`**. The Espressif developer, @xcguang, uses `master` in much the same way we use `dev`, and as at time of posting its last commit was on 23 Apr.  This build is titled **3.1.0-dev**   
-  **`RELEASE=latest-3.0`**. xcguang also regularly cherry picks relevant commits from master back into the `release-v3.0.0` branch to fix known issues with 3.0.  The last was also on 23rd Apr, and this branch is almost 100 commits ahead of the v3.0 tagged release.  This build is titled **SDK 3.0.1-dev**.

I haven't done exhaustive testing, but the firmware builds based on `master` and `latest-3.0` seem stable, and certainly don't manifest #2751.  Since Espressif are using Github, there is nothing to stop us locking our `master` and `dev` branches to specific commits on, say, the SDK 3.0.1-dev branch, so that given NodeMCU firmware builds are well determined.

My suggestion is that we switch `dev` to using a specific commit of the Espressif SDK 3.0.1-dev branch and defer a new update of master to a SDK 3.x-based release until we can take a firm view of its stability.  This is almost a policy issue so I would welcome the views of the other commtters: @devsaurus @HHHartmann @jmattsson @marcelstoer @nwf @pjsg.

pjsg, 2019-05-13:
I think that we should switch to a specific commit of their 3.0.1-dev branch for our `dev`. However, I'm much less confident about what we should do for the next master drop. I think, that as the time gets close, we need to evaluate what version we should lock to for our `master`.

jmattsson, 2019-05-14:
Sounds sensible to me.

TerryE, 2019-05-14:
I have updated the Makefile for `dev` to  use the latest **releasev3.0.0** commit as the default SDK for the `dev` make: espressif/ESP8266_NONOS_SDK@e4434aa.  This fixes #2751 and with the 2½ Kb of data moved from dRAM to iRAM we've got as much dRAM available with the v2.2.1 builds.  There's about 12Kb of iRAM also free, so if anyone can come up with a way of using this then _workable_ suggestions would be appreciated.

PR #2757 (2019-05-17): SDK-3.0 tranche updates
Fixes #2729, #2743,  #2751 and #2754 

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [ ] The code changes are reflected in the documentation at `docs/*`.  _Not applicable_

### Changes in this PR

-  `Makefile` now supports different SDK versions as per #2729.  The default is the latest SDK 3.0.1-dev (e4434aa) version.
-  `app/Ma [... cut]
closed 2019-05-17 by TerryE