nodemcu/nodemcu-firmware #2744: [question]how can I use the Lua Modules

hanfengcan, 2019-05-04:
HI, I build the firmware by docker. there are some `Lua Modules` that I can find in the document. 

eg:
[image]

when I try to use this modules, the prompt appears

```
httpServer.lua:1: module 'httpserver' not found:
	no field package.preload['httpserver']
	no file 'httpserver.lc'
	no file 'httpserver.lua'
stack traceback:
	[C]: in function 'require'
	httpServer.lua:1: in main chunk
	[C]: in function 'dofile'
	stdin:1: in main chunk
```

I read the section about **building the firmware**, but not found about the lua modules.

Is it the lua modules not built into the firmware?

Is it need to update by myself when I want to use the lua modules?

chilipeppr, 2019-05-04:
You have to upload your modules to your NodeMCU after you connect to it via
serial port after you've installed the firmware. Once you connect over
serial you get a Lua command prompt and then you can start uploading Lua
code and run it. I use chilipeppr.com/esp32 or chilipeppr.com/nodemcu for
ESP8266 to do this.

hanfengcan, 2019-05-04:
@chilipeppr thanks, it dose work. 

but, I have another question.  I try the example `http-example.lua`, the callback does not response.

and 
```
server:getaddr()
80	0.0.0.0
```
What should I pay attention to when using this module?

TerryE, 2019-05-04:
Hi @hanfengcan, this issues list isn't the right place to ask "how to" questions. See #1010 for more details.

closed 2019-05-04 by TerryE

hanfengcan, 2019-05-05:
@TerryE I know, thanks