nodemcu/nodemcu-firmware #2749: file.read(n) reads always 128 bytes

shimosaurus, 2019-05-07:
### Expected behavior
As [example](https://nodemcu.readthedocs.io/en/dev/modules/file/#example-object-model_1)  in the doc.
```Lua
-- print the first 5 bytes of 'init.lua'
fd = file.open("init.lua", "r")
if fd then
  print(fd:read(5))
  fd:close(); fd = nil
end
```
### Actual behavior
Reads always 128 bytes

```
> print(#fd:read(5))
128
> print(#fd:read(50))
128
```

### Test code
```Lua
fd = file.open("init.lua", "r")
if fd then
  print(#fd:read(5))
  fd:close(); fd = nil
end
```
returns `128`
### NodeMCU version
Breaks in commit [c3e24436f2968b8610275fce9b28e3b32588562b](https://github.com/nodemcu/nodemcu-firmware/commit/c3e24436f2968b8610275fce9b28e3b32588562b)
 `Lua string optimisation in file.c + get/put contents methods (#2717)`

### Hardware
[Lolin D1 mini Pro V2.0.0](https://wiki.wemos.cc/products:d1:d1_mini_pro#d1_mini_pro)

TerryE, 2019-05-07:
@shimosaurus, looks like a good catch.  I will check this out.

TerryE, 2019-05-07:
Sorry, the logic was flipped on `app/modules/file.c:L435`. I've just put in a fix to correct this which I have merged in now that the Travis check has completed.

PR #2750 (2019-05-07): Fix 2749 + restore correct user_modules.h
Fixes #2749

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

One-line fix  to issue, plus backing out incorrect replacement of `app/include/user_modules.h`
closed 2019-05-07 by TerryE