nodemcu/nodemcu-firmware #2754: FIrmware fails to build with #define PMSLEEP_ENABLE

pjsg, 2019-05-11:
### Expected behavior
The firmware should build
### Actual behavior
It fails building app/modules/node.c

### NodeMCU version
Today's dev branch

The solution is to remove the extra comma......

TerryE, 2019-05-11:
> Even then, I thought that the LROT stuff had to have the keys in alphabetical order -- which is vry much not the case for node.c

There is a trade-off here: you need a minimum of maybe 6 entries in your ROTable before a binary chop is on average any faster than a simple linear search -- and the search only takes place now anyway where there is a cache miss -- for example where `__index` meta is used to chain ROTables.  So my current thinking is to make the binary chop optional, and only use it initially on the higher frequency long tables where this is the case -- for example the baselib ROTable.

As far as the comma goes -- Philip Is can do this on my next PR if you want.  Your call.

pjsg, 2019-05-11:
Do it on your next pr

TerryE, 2019-05-11:
Philip, thanks for the feedback and will do :smile:

TerryE, 2019-05-16:
#2757 fixes this.

closed 2019-05-16 by TerryE