nodemcu/nodemcu-firmware #2756: Cannot connect to NodeMCU Amica ESP8826 after firmware flash

nolash, 2019-05-13:
* **Operating System**  
 
Linux sostenuto 5.0.8-arch1-1-nousbserial #1 SMP PREEMPT Fri May 10 00:18:58 CEST 2019 x86_64 GNU/Linux

* **ESP Device/Revision**  

NodeMCU Amica. Says on the back use baud rate `9600` and install driver `cp2102`

```
$ lsusb -v -d 10c4:ea60

Bus 002 Device 071: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP2102/CP2109 UART Bridge Controller [CP210x family]
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x10c4 Cygnal Integrated Products, Inc.
  idProduct          0xea60 CP2102/CP2109 UART Bridge Controller [CP210x family]
  bcdDevice            1.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              2 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
```

* **NodeMCU-Tool Version**  
  Please use the least recent version to validate the issue  
  `3.0.2`
* **Node.js Version**  

`v11.15.0`

* **NodeMCU LUA Firmware Version**  
  `1.5.4`

-------------

## Debug Output ##

```

$ sudo esptool --port /dev/ttyUSB0 --baud 115200 write_flash -fm dio 0x0000 /home/lash/Downloads/nodemcu-1.5.4.1-final-15-modules-2019-05-13-18-36-43-float.bin 
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 18:fe:34:f1:f5:10
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0240
Compressed 433556 bytes to 277352...
Wrote 433556 bytes (277352 compressed) at 0x00000000 in 24.5 seconds (effective 141.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

```

The blue light on the `ESP8266` blinks briefly after power on, then stops. When I connect with `screen /dev/ttyUSB0` and try `AT` there is no echo and no response.

```
$ lsmod | grep usb
btusb                  57344  0
btrtl                  20480  1 btusb
btbcm                  16384  1 btusb
btintel                28672  1 btusb
bluetooth             651264  4 btrtl,btintel,btbcm,btusb
usbserial              57344  1 cp210x
usb_storage            73728  1 uas
usbhid                 61440  1 wacom
hid                   147456  3 wacom,usbhid,hid_generic
scsi_mod              253952  5 sd_mod,usb_storage,uas,libata,sg

$ dmesg
[...]
[ 1062.388726] usb 3-1: new full-speed USB device number 5 using xhci_hcd
[ 1062.530697] usb 3-1: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[ 1062.530704] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1062.530708] usb 3-1: Product: CP2102 USB to UART Bridge Controller
[ 1062.530711] usb 3-1: Manufacturer: Silicon Labs
[ 1062.530713] usb 3-1: SerialNumber: 0001
[ 1062.532112] cp210x 3-1:1.0: cp210x converter detected
[ 1062.533955] usb 3-1: cp210x converter now attached to ttyUSB0

```

nwf, 2019-05-13:
This is the wrong venue for such questions (see #1010), and you appear perhaps a little lost, I'm afraid.  That said...

You should not expect to speak the Hayes AT command set to NodeMCU; it will give you a Lua interpreter over serial and the usual prompt is `> `.  Please carefully review https://nodemcu.readthedocs.io/en/latest/flash/#upgrading-firmware and check your terminal baud rate; despite what the module says on its back, NodeMCU firmware speaks at 115200 by default.

As an aside, 1.5.4 is an ancient build and, unless you're confined to 512K FLASH modules, you will probably be better served by master.

closed 2019-05-13 by nwf

nolash, 2019-05-14:
@nwf Thanks. Indeed, between master, the default data in the SDK and re-trying `nodemcu-tools` I got it working. And yes, seems `/dev/ttyUSB0` doesn't offer UART out of the box, but rather lua interactive terminal..?

HHHartmann, 2019-05-14:
If you are looking for AT commands this is the wrong place. 
You need to flash a different firmware.