nodemcu/nodemcu-firmware #2758: Running wifi.setmode(wifi.NULLMODE) will erase saved credentials.

KT819GM, 2019-05-18:
### Expected behavior
Running wifi.setmode(wifi.NULLMODE) does not erase WiFi credentials from flash.
### Actual behavior
Running wifi.setmode(wifi.NULLMODE) does erase WiFi credentials from flash. By default  LFS.HTTP_OTA example will run this command and will make life more complicated for people who use wifi.startsmart(). 
### Test code

```Lua
wifi.setmode(wifi.NULLMODE)
```
### NodeMCU version
2.2.1-master_20190405

### Hardware
ESP-07S generic module

galjonsfigur, 2019-05-18:
According to documentation, there is an optional parameter in `wifi.setmode()` function to chose whether or not to save wifi mode to flash. Can you try changing line:
`wifi.setmode(wifi.NULLMODE)`
to
`wifi.setmode(wifi.NULLMODE, false)`
in https://github.com/nodemcu/nodemcu-firmware/blob/master/lua_examples/lfs/HTTP_OTA.lua#L68
and report if your code works?

KT819GM, 2019-05-18:
Thank you, with `wifi.setmode(wifi.NULLMODE, false)` behaviour is as I expected. In my opinion it should be default mode, as its primary purpose is to set modem to low power state.

My apologies for not paying attention into documentation as I had to.

closed 2019-05-18 by KT819GM

galjonsfigur, 2019-05-18:
You could also send a Pull Request to fix this behavior in HTTP_OTA - I've never had a chance to use it, but I suppose it can be frustrating.

PR #2778 (2019-06-01): Modify HTTP OTA to not erase saved credentials (#2758)
Fixes #2758

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

The author haven't raised a PR related to this issue but I think it would be a good idea to add this change.