nodemcu/nodemcu-firmware #2759: =node.info() error on esp32

coolwolf, 2019-05-18:
on esp32
**=node.info()** command
returns:
**stdin:1: attempt to call field 'info' (a nil value)**

marcelstoer, 2019-05-19:
Yep, [ain't implemented yet](https://github.com/nodemcu/nodemcu-firmware/blob/dev-esp32/components/modules/node.c#L393). Now you can argue whether it's a "bug" that it's included in the API documentation or that it's not implemented 😉 

One would need to port https://github.com/nodemcu/nodemcu-firmware/blob/4905381c004bdeaf744a60bb940c1906c2e052d4/app/modules/node.c#L125-L136 to ESP32.

stuartpb, 2019-06-10:
For that matter, what about a field in node.info's returns that explicitly says whether the current chip is an ESP32 vs ESP8266, since there are a lot of divergent APIs between the two?

ildar, 2019-08-19:
This is the current way to determine then )))

marcelstoer, 2020-03-22:
@chilipeppr maybe something for you to tackle?

chilipeppr, 2020-03-22:
That last few months I've had to dive back hard into the day job, so ruined
my ability to do much. Sorry. The tough nature of open source software and
real life.

PR #3655 (2024-07-31): Added node.info()
Fixes #2759

- [x] This PR is for the `dev-esp32` branch rather than for the `release` branch.
- [x] This PR is compliant with the [other contributing guidelines](/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

This PR adds support for the previously missing node.info() method on esp32. 

Test  code:
```
function printTable(t) for k, v in pairs(t) do print(k, v) end end

print("\n====== no params:") printTable(node.info())
print("\n====== build conf [... cut]
closed 2024-07-31 by marcelstoer