nodemcu/nodemcu-firmware #2764: lfs_fragments.lua: replace old static timer

alistairwitt, 2019-05-21:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Timer operates
### Actual behavior
Error as follows:
init.lua:1: attempt to call field 'alarm' (a nil value)
stack traceback:
	runlfs.lua:1: in main chunk
	[C]: in function 'dofile'
	stdin:1: in main chunk

### Test code
nodemcu-firmware/lua_examples/lfs/lfs_fragments.lua
lines 59 to 62 are using the old/removed static timer and need to be replaced with a dynamic one.

**Suggested replacement code:**
local initTimer = tmr.create()
```
initTimer:register(1000, tmr.ALARM_SINGLE,
    function()
        local fi=node.flashindex; return pcall(fi and fi'_init')
    end
    )
initTimer:start()
```

### NodeMCU version
master or dev branch

### Hardware
ESP8266, Wemos D1 Mini

8<------------------------ END BUG REPORT -------------------------------------

marcelstoer, 2019-05-21:
Was recently fixed with #2752.

alistairwitt, 2019-05-21:
I can only see changes in #2752 that relate to luaOTA, not lfs_fragments.lua. Am I missing something?
Cheers,
Alistair

marcelstoer, 2019-05-21:
Argh...I spoke too soon, sorry 🤦‍♂

HHHartmann, 2019-05-21:
@alistairwitt your suggested changes look good. 
Do you want to create a PR?

alistairwitt, 2019-05-21:
@HHHartmann I'll give it a go. I should be able to get to this around 12 hours from now.

PR #2765 (2019-05-22): Replaced obsolete static timers in lfs_fragments.lua (nodemcu #2764)
Fixes #2764.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

Replaces the obsolete static timers with dynamic ones.
closed 2019-05-22 by alistairwitt