nodemcu/nodemcu-firmware #2777: Socket IP available in callback?

cwrseck, 2019-05-31:
I've been trying to work around the absence of NodeMCU ping(); net.dns.resolve()
is unreliable due to DNS caching.  The hardware is a Wemos D1, and the software
the 2019-04-05 NodeMCU release.
```
The code:

    local tcpskt = net.createConnection(net.netTCP, 0)
    if tcpskt ~= nil
    then
        print("good tcpskt")
        tcpskt:on("receive", function(sk, c) print(c) end)
        tcpskt:on("connection",
            function(sk, ip) local str = ip or "no IP" ; print("TCP socket " .. str) ; sk:close() end)
        tcpskt:connect(53, "8.8.8.8")
    else
        print("bad tcpskt")
    end
```
produces the strings "good tcpskt" and "TCP socket no IP".  If the IP is
changed to "8.8.8.1", whose port 53 is closed, it produces only the
"good tcpskt" string, as expected.

Is this expected behaviour, ie: that the IP is uninitialised in the callback?
It isn't documented.

C W Rose

galjonsfigur, 2019-05-31:
According to the [code](https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/modules/net.c#L153), on `"connection"` callback there is only one value in callback function and that is socket object.

As for missing `ping` function - you can do something like this:
```lua
rtctime.set(0, 0)

function ping(ip, port)
  local startTime = select(2, rtctime.get())
  local conn = net.createConnection(net.TCP, 0)
  conn:on("connection", function(socket)
    endTime = select(2, rtctime.get())
    deltaTime = endTime - startTime
    if deltaTime < 0 then deltaTime = 1000000 - startTime + endTime end
    print("IP: " .. ip .. " is reachable, time: " .. deltaTime/1000 .. "ms")
  end)
  conn:connect(port, ip)
end

ping("8.8.8.8", 53)
```

It's not as reliable as native lwIP function that is not implemented (there was an attempt: #1555) but should work.

nwf, 2019-05-31:
Note that, given a socket `s`, I think `s:getpeer()` will return the remote IP address (dotted quad string) and port (integer).

cwrseck, 2019-06-01:
@galjonsfigur - so it looks as if the callback is producing a result, just not the result I expected.  The ping() code is much nicer than the rather clumsy code I'm using, so I'll try that.  Many thanks.

C W Rose

closed 2019-06-01 by nwf