nodemcu/nodemcu-firmware #2779: dev-esp32 firmware not building with bluetooth modules

sahilanguralla, 2019-06-02:
I was trying to build firmware for ESP32, it built successfully when I was not adding bluetooth module. But when I add bluetooth it fails to build.

### Expected behavior
Should compile successfully

### Actual behavior
```
make[2]: Leaving directory '/home/sahil/learner/nodemcu-firmware/build/wpa_supplicant'
make[2]: Entering directory '/home/sahil/learner/nodemcu-firmware/build/xtensa-debug-module'
CC build/xtensa-debug-module/trax.o
AR build/xtensa-debug-module/libxtensa-debug-module.a
make[2]: Leaving directory '/home/sahil/learner/nodemcu-firmware/build/xtensa-debug-module'
LD build/NodeMCU.elf
/home/sahil/learner/nodemcu-firmware/build/base_nodemcu/libbase_nodemcu.a(user_main.o): In function `esp_event_send':
/home/sahil/learner/nodemcu-firmware/components/base_nodemcu/user_main.c:42: multiple definition of `esp_event_send'
/home/sahil/learner/nodemcu-firmware/build/esp32/libesp32.a(event_loop.o):/home/sahil/learner/nodemcu-firmware/sdk/esp32-esp-idf/components/esp32/event_loop.c:75: first defined here
collect2: error: ld returned 1 exit status
/home/sahil/learner/nodemcu-firmware/sdk/esp32-esp-idf/make/project.mk:460: recipe for target '/home/sahil/learner/nodemcu-firmware/build/NodeMCU.elf' failed
make[1]: *** [/home/sahil/learner/nodemcu-firmware/build/NodeMCU.elf] Error 1
make[1]: Leaving directory '/home/sahil/learner/nodemcu-firmware'
Makefile:18: recipe for target 'all' failed
make: *** [all] Error 2
```
### NodeMCU version
branch: dev-esp32
commit: e30904b60760f8c290ada6deb2a44480ab884ba2

### SDK Config
[sdkconfig.txt](https://github.com/nodemcu/nodemcu-firmware/files/3244644/sdkconfig.txt)

pyr0ball, 2019-06-05:
+1 having the same issue on Ubuntu 18.04 build environment

loadedtech, 2019-07-13:
I'm getting the same error.  
(It builds fine without the NodeMCU bluetooth module, although I need to erase the sdkconfig file and re-run configure-esp32 without enabling bluetooth to get the build to complete)

ildar, 2019-08-13:
no fresh ideas here? I stumbled too.

loadedtech, 2019-08-13:
FYI I no longer have this problem ... now I can't build with or without Bluetooth :-(

@pyr0ball are you using the latest docker image and branch of dev-esp32?

pyr0ball, 2019-08-13:
not docker, I'm running on bare metal

ildar, 2019-08-13:
IMHO the problem is that NodeMCU redefines a function (esp_event_send)
which is a part of SDK. This is done successfully with most modules but BT
module gets SDK function instance and brings it with it. Linker complains.

hanhealthy, 2019-10-31:
is there any updates or solution to use BLE with nodemcu?

Engineer-of-Stuff, 2019-11-19:
bump. I'd really like bluetooth.

wbbw55, 2019-12-19:
i have the same error, i like bt :(

girishkr, 2020-03-10:
same error 
nodemcu-firmware-esp32/components/base_nodemcu/user_main.c:42: multiple definition of `esp_event_send'

landswipe, 2020-04-24:
I ran into the same issue, any solutions? The reason you had to clear sdkconfig is likely because the root cause is in "CONFIG_BT_ENABLED" which is selected by the bthci component.

landswipe, 2020-04-24:
I managed to hack some changes to get it working.

If you just rename esp_event_send in the sdk under components/esp32/event_loop.c, it will work. In my case I had to ensure the BTDM mode was set (I just accepted the defaults from there). It looks like there is still very rudimentary support even at the component selection side, but there is enough to get the basics working.

I actually only need rudimentary BLE scanning for my application, so this will hopefully do it. It would be nice to build out the API based on the BT samples for gatt/discovery support too.

marcelstoer, 2020-04-25:
I'm wondering if a change in the underlying ESP-IDF is causing this. Something that has changed since the time @jmattsson donated the BT module.

jmattsson, 2020-04-26:
Sorry, I haven't been active much in the nodemcu community the last year due to lack of time.
Could you please try this patch? I don't have hardware on hand at the moment (COVID-19 lockdown).

```
diff --git i/components/base_nodemcu/user_main.c w/components/base_nodemcu/user_main.c
index d18515dd..d2982a49 100644
--- i/components/base_nodemcu/user_main.c
+++ w/components/base_nodemcu/user_main.c
@@ -16,6 +16,7 @@
 #include "sdkconfig.h"
 #include "esp_system.h"
 #include "esp_event.h"
+#include "esp_event_loop.h"
 #include "nvs_flash.h"
 #include "flash_api.h"
 
@@ -35,16 +36,16 @@
 static task_handle_t esp_event_task;
 static QueueHandle_t esp_event_queue;
 
-// We provide our own esp_event_send which hooks into the NodeMCU task
-// task framework, and ensures all events are handled in the same context
-// as the LVM, making life as easy as possible for us.
-esp_err_t esp_event_send (system_event_t *event)
+// The callback happens in the wrong task context, so we bounce the event
+// into our own queue so they all get handled in the same context as the
+// LVM, making life as easy as possible for us.
+esp_err_t bounce_events(void *ignored, system_event_t *event)
 {
   if (!event)
     return ESP_ERR_INVALID_ARG;
 
   if (!esp_event_task || !esp_event_queue)
-    return ESP_ERR_INVALID_STATE; // too early!
+    return ESP_ERR_INVALID_STATE; // too early (should be impossible)!
 
   portBASE_TYPE ret = xQueueSendToBack (esp_event_queue, event, 0);
   if (ret != pdPASS)
@@ -152,6 +153,8 @@ void app_main (void)
     xQueueCreate (CONFIG_SYSTEM_EVENT_QUEUE_SIZE, sizeof (system_event_t));
   esp_event_task = task_get_id (handle_esp_event);
 
+  esp_event_loop_set_cb(bounce_events, NULL);
+
   input_task = task_get_id (handle_input);
 
   ConsoleSetup_t cfg;
```

landswipe, 2020-04-26:
Thanks @jmattsson, I get "E (1214) event: Event loop not initialized via esp_event_loop_init, but esp_event_send called" on boot...

jmattsson, 2020-04-27:
Thanks @landswipe. Let's try this one instead then:
```
diff --git i/components/base_nodemcu/user_main.c w/components/base_nodemcu/user_main.c
index d18515dd..d2982a49 100644
--- i/components/base_nodemcu/user_main.c
+++ w/components/base_nodemcu/user_main.c
@@ -16,6 +16,7 @@
 #include "sdkconfig.h"
 #include "esp_system.h"
 #include "esp_event.h"
+#include "esp_event_loop.h"
 #include "nvs_flash.h"
 #include "flash_api.h"
 
@@ -35,16 +36,16 @@
 static task_handle_t esp_event_task;
 static QueueHandle_t esp_event_queue;
 
-// We provide our own esp_event_send which hooks into the NodeMCU task
-// task framework, and ensures all events are handled in the same context
-// as the LVM, making life as easy as possible for us.
-esp_err_t esp_event_send (system_event_t *event)
+// The callback happens in the wrong task context, so we bounce the event
+// into our own queue so they all get handled in the same context as the
+// LVM, making life as easy as possible for us.
+esp_err_t bounce_events(void *ignored, system_event_t *event)
 {
   if (!event)
     return ESP_ERR_INVALID_ARG;
 
   if (!esp_event_task || !esp_event_queue)
-    return ESP_ERR_INVALID_STATE; // too early!
+    return ESP_ERR_INVALID_STATE; // too early (should be impossible)!
 
   portBASE_TYPE ret = xQueueSendToBack (esp_event_queue, event, 0);
   if (ret != pdPASS)
@@ -152,6 +153,8 @@ void app_main (void)
     xQueueCreate (CONFIG_SYSTEM_EVENT_QUEUE_SIZE, sizeof (system_event_t));
   esp_event_task = task_get_id (handle_esp_event);
 
+  esp_event_loop_init(bounce_events, NULL);
+
   input_task = task_get_id (handle_input);
 
   ConsoleSetup_t cfg;
```

landswipe, 2020-04-27:
Yeah I had actually tried this, but it just triggers the internal watchdog task. 

I am still trying to trace this as I am green with RTOS. I can see the wifi change events for example hitting esp_event_send and then it calls the bounce event handler.

Happy to keep trying! :)

landswipe, 2020-04-27:
The events get passed through to evregs->callback (&evt); in handle_esp_event() but don't look like they're calling back to lua...

landswipe, 2020-04-28:
@jmattsson I managed to get it working by removing the superfluous esp_event_process_default() call in user_main.c's handle_esp_event() function. I just noticed that event_loop.c was already calling it, but I am not sure if that is a valid fix. Why it broke the wdt I am not sure, reset events still arrived but it seemed less in comparison. I suspect it is because it is now just bouncing the events into its own queue now and processing them in task_pump_messages() rather than trying to override/simulate parts of event_loop.

I am not aware of the history and how all this came to be with the ESP8266 implementation, but I assume there were a lot of backward compatibility changes when they shifted to freertos. 

Still learning... :)

jmattsson, 2020-04-28:
@landswipe Ah, good catch! Yes, that could certainly do it, since in effect the SDK libs ended up seeing each event twice, which no doubt was very confusing to it.
Do you want to raise a PR with your working fix? Since my track record on this one is "not very good"? :D

landswipe, 2020-04-29:
@jmattsson Will do, I have a couple of PRs I will send shortly :) Thanks for the help on this one!

landswipe, 2020-05-01:
I was going to post the PR, but noticed a problem with just basic scanning. It looks like there are so many BLE advertisement reports that the watch dog timer kicks in after about 5 seconds or so. If I remove lua_call in handle_hci_event() the watch dog does not trigger.

@jmattsson would you be able to point me in the right direction to resolve this?

pjsg, 2020-05-01:
I had a similar problem with the wifi monitor stuff -- I ended up adding a filtering layer so that you didn't get the callback on every management frame: https://nodemcu.readthedocs.io/en/master/modules/wifi_monitor/#wifimonitorstart

Also, I had a *lot* of attributes that could be read out of the packet object -- I ended up just have a single __index method on the packet object that handled all the fields. It was much cheaper to do this than to create the 200 field values whenever a new packet object was created.

Check app/modules/wifi_monitor.c for my approach.

landswipe, 2020-05-02:
FYI, I just migrated the idf sdk to branch "release/v3.3" and the watch dog issue has disappeared. Thanks for the heads up @pjsg

marcelstoer, 2020-05-02:
> I just migrated the idf sdk to branch "release/v3.3" and the watch dog issue has disappeared

We're currently on ESP-IDF 3.3.1 but it appears https://github.com/espressif/esp-idf/releases/tag/v3.3.2 fixes a number of Bluetooth issues. Looks like we should bump our version.

marcelstoer, 2020-05-06:
@landswipe did you test against v3.3.2 as well or just against the latest release/v3.3?

landswipe, 2020-05-06:
Originally I tested against release/v3.3 but then dropped back to tags/v3.3.2 (both are fine). I have a pull request almost ready, but found some other issues I am working through. Obviously the fix for this issue will also require the sdk to be bumped...

marcelstoer, 2020-05-06:
>  Obviously the fix for this issue will also require the sdk to be bumped...

Ideally we bump this in a separate PR for better visibility of the changes.

BenjaminWongCK, 2020-06-17:
When I went through the NodeMCU documentation, I found a **Bluetooth** module called **bthci** under C Module of the documentation; however, when I tried to do a  "NodeMCU custom builds" on Web for an newly purchased ESP32 DevKit, I was unable to find the btchi option from dev-esp32 [BETA] build selection.   could anyone enlighten me how to include the btchi module to an  NodeMCU custom builds firmware.

marcelstoer, 2020-06-17:
> could anyone enlighten me how to include the btchi module to an NodeMCU custom builds firmware.

You can't. It won't make sense to make it available there if it's broken i.e. as long as this issue is not resolved.

BenjaminWongCK, 2020-06-17:
> > could anyone enlighten me how to include the btchi module to an NodeMCU custom builds firmware.
> 
> You can't. It won't make sense to make it available there if it's broken i.e. as long as this issue is not resolved.

Understood, hope the broken can be fixed very soon.

landswipe, 2020-06-18:
Hey, sorry I have been busy with other things, but haven't forgot about this pull request. 

I do have it working now, but there are some compromises. There are still an underlying issue with feeding a lot of hci data through lua callbacks stalling the watchdog, but I have been able to work around most of it to a point where it works for my application.

marcelstoer, 2020-06-20:
>>  Obviously the fix for this issue will also require the sdk to be bumped...
>
> Ideally we bump this in a separate PR for better visibility of the changes.

I prepared #3173 for that.

thornley-david, 2020-07-05:
Once the upgrade to 3.3.2 LTS is ready, I'll submit a pull request. Changes are currently here https://github.com/thornley-david/nodemcu-firmware/commit/ed5bef455403c441c7a8a10b873e3aba09060665 against chore/bump-esp-idf if you want to take a look.

marcelstoer, 2020-07-05:
Thanks to you the IDF bump is ready at #3196 but I don't like merging my own PRs. I'll for a few days to see if one of the other maintainers will take a look.

thornley-david, 2020-07-08:
@marcelstoer Great! A minor contribution on my part. Hopefully this gets through now so others can derive some benefit. Many thanks for your help!

PR #3201 (2020-08-03): Fix bthci compilation issue
Fixes #2779

- [X] This PR is for the `dev` branch rather than for `master`.
- [X] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [ ] I have thoroughly tested my contribution.
- [ ] The code changes are reflected in the documentation at `docs/*`.

Manifests as multiple definition of esp_event_send during compilation, as bthci triggers inclusion of event_loop.c

Kudos to @jmattsson for suggestion to adjust user_main to properly register a hand [... cut]
closed 2020-08-03 by marcelstoer