nodemcu/nodemcu-firmware #2780: xpcall() udpskt panic

cwrseck, 2019-06-02:
Given the current release NodeMCU running on a Wemos D1, the code:
```
    local bcast = wifi.sta.getbroadcast()
    bcast = bcast or "192.168.1.255"
    local rc, udpskt = pcall(net.createUDPSocket)
    if rc
    then
        local str = "Message"
        udpskt:on("sent", function(s, c) s:close() end)
        local r, e = xpcall(udpskt:send(UDPPort, bcast, str),
            function(e) print("state_f: xpcall() " .. e) return end) -- 576
        -- Remove the colon operator.
        -- local e = xpcall(udpskt.send(udpskt, UDPPort, bcast, str), xperr)
    end
```
produces the error message:
```
PANIC: unprotected error in call to Lua API (switch.lua:576: routing problem)
```
A standard pcall() has the same result.  The failure occurs intermittently,
which makes me think that it's a race condition, and Issue 2161 may apply,
although the wireless remains available up to the crash.  Removing the colon
operator makes no difference.

I doubt I can debug the race condition, and the widget involved needs to run
continuously.  Is there any known fix for the panic, or do I just have to drop
the use of udpskt?

C W Rose

nwf, 2019-06-02:
Similar to what I said in #2161, that's **not** how `pcall` or `xpcall` works. You're causing Lua to evaluate the expression and then passing the result to `xpcall`, which is not what you meant.  Try reading https://stackoverflow.com/questions/30125726/how-to-use-xpcall-with-a-function-which-has-parameters

galjonsfigur, 2019-06-02:
According to the [Lua 5.1 Manual](https://www.lua.org/manual/5.1/manual.html#pdf-xpcall) `xpcall` takes only two arguments: function and error handler. There is no way to call a function with parameters other than something like this:
```lua
local e = xpcall(function() udpskt.send(udpskt, UDPPort, bcast, str) end, xperr)
```
I was having similar problem some time ago. `pcall` in your code is just used in the wrong way. A good example:
```lua
local function errHandler()
  print("Failed")
end

err = xpcall(print("Hello"..nil), errHandler)
-- Will return:
-- stdin:1: attempt to concatenate a nil value
-- stack traceback:
--         stdin:1: in main chunk

err = xpcall(function() print("Hello"..nil) end, errHandler)
-- Will return:
-- Failed
```
`routing problem` means that there is no route to the destination - and that can mean for example that there is no connection to the Internet but only to the gateway. It can happen and that's what `pcall` and `xpcall` is for.

EDIT:
@nwf was faster but I still will leave this comment - maybe someone will find it useful.

cwrseck, 2019-06-02:
@nwf, @galjonsfigur Yes, well, it helps if you read the manual; in this case
Section 8.4  of "Programming in Lua".  I've managed to read that section
several times in the last couple of days  without, apparently, understanding
it at all.  Thanks for the help, and sorry for the noise (and both comments
are pretty useful).

C W Rose

TerryE, 2019-06-04:
> According to the [Lua 5.1 Manual](https://www.lua.org/manual/5.1/manual.html#pdf-xpcall) `xpcall` takes only two arguments: function and error handler. There is no way to call a function with [...]
There is a way to pass arguments -- albeit a cludge -- and that is because _upvals_ are available within the context of the closure, for example
```Lua
do
  local sk, port, bc, s = -- ... whatever they take
  local e = xpcall(function() sk:send(p, bc, s) end, xperr)
end
```
Of course you could also use the bare variables, and in the case of collectible objects like `sk` which are passed by reference then there is no good reason not to.  For strings, ordinals, etc. you might want to snapshot them into a local variable as above as upvals are references and these local copies are assigned and thus by value.

closed 2019-11-09 by TerryE