nodemcu/nodemcu-firmware #2781: MQTT connect fail with new LFS release

chipec84, 2019-06-02:
Hello,
I am trying to connect to mqtt after wifi properly connected but I get a connect fail reason -5 (There is no broker listening at the specified IP Address and Port). 
I building the firmware using docker and I am using LFS.

I get this error with the latest release only "2.2.1-master 20190405" but NOT with the previous one "2.2.1-master 20181207".

I have checked the documentation and it seems that nothing has changed regarding this topic.
Any idea of what I could be doing wrong? (missing something when building firmware? something wrong in my code?)

local ipBoker = "serverName"
local portBroker = "8883"
m:connect(ipBroker, portBroker, 1,
	function(client) 
		print("MQTT => connected...")
		subscribe()
	end, 
	function(client, reason) 
		print("MQTT => connect failed, "..reason.." wait...")
		m:close()
        end
end)

Thank you.

galjonsfigur, 2019-06-02:
I've tested it on slightly modified code on both `dev` and latest `master`:
```lua
ipBroker = "test.mosquitto.org"
portBroker = "8883"
m = mqtt.Client("TestClientID", 120)
m:connect(ipBroker, portBroker, 1,
  function(client)
    print("MQTT => connected...")
  end,
  function(client, reason)
    print("MQTT => connect failed, "..reason.." wait...")
    m:close()
  end
)

-- Result:
-- MQTT => connected...
```
And everything seems to work. I don't know if your code is only an example or a fragment of something bigger but there were a few typos: `ipBoker` instead of `ipBroker`and one `end` too much.

Error -5 could be caused by not enough RAM available - you would have to check for any `E:M` or use `print(node.heap())` to check it. Also if you can, build the firmware with `DEVELOP_VERSION` uncommented [here](https://github.com/nodemcu/nodemcu-firmware/blob/master/app/include/user_config.h#L184) and look for any information.

chipec84, 2019-06-02:
Thank you
The typo is a wrong copy/typing, likewise for the extra end, this piece of code is indeed a small fragment from a bigger project. although I have an SSL connection, the heap is at 20500, because I am using LFS for 95% of the code. I call my connect function from the LFS. 
I tried the previous two releases 2.2.1-master_20181207 and 2.2.1-master_20180915, they both work fine. I will try to build with DEVELOP_VERSION.

dtran123, 2019-06-03:
MQTT works fine for me with LFS on latest dev build. MQTT works also with certificates and CA verification.

nwf, 2019-06-03:
@chipec84 You may be running afoul of #2587, merged as 6e95d74fbdfc40f4ed587871942ebc97b9dd23d9, which changed the TLS vocabulary that NodeMCU speaks.  Your broker may, for example, be speaking only TLS 1.0 or non-PFS ciphersuites; it's also possible, but unlikely, that you're bumping into the reduced TLS 4K buffer size if the certificate chain for your broker is substantial.

You will probably be best served by reviewing your broker's configuration and/or its logs and/or obtaining a packet trace of NodeMCU attempting to establish a connection with said broker.

dtran123, 2019-06-03:
Enabling TLS debug traces would certainly help to find out if things such as max msg length has exceeded (I didn't mention this in previous post that I had to increase this value to 5009+ to accommodate for requirements of the TLS certs negotiation), etc. Without these traces, I am usually in the dark as to what is causing the error.

closed 2019-06-04 by marcelstoer