nodemcu/nodemcu-firmware #2782: Simpler way to implement control via Bluetooth SPP or alternative interfaces?

Alfrederson, 2019-06-04:
I kind of managed to implement using the built-in bluetooth interface by copying arduino bluetooth serial library (https://github.com/espressif/arduino-esp32/tree/master/libraries/BluetoothSerial) into my own component. I initialize it calling a function that is analogous to BluetoothSerial.begin(), but I had to change some things in the nodemcu original sourcecode for ESP32.

Namely, I had add a line to the console_getc in console.c to set the "*c" argument to a char read from the bluetooth spp's queue if there was nothing in the uart0Q queue. I also had to make the Bluetooth SPP calback call user_process_input (passed as a function pointer) after a ESP_SPP_DATA_IND_EVT event. It almost works in the sense that I can send print("something") from a bluetooth serial terminal app and the ESP32 will print "something" in the serial terminal I'm running on my computer, to which the esp32 is connected via a usb-serial device. However, the esp32 enters an infinite loop if I use send a file.open() from my android phone so that I could, for example, program it from a mobile IDE.

Now, this is a really hackish approach and I'm thinking that there should be  a way to generate input from within the firmware itself. I.e: there should be an interface one could use to send commands to the lua interpreter and to write output to through a callback function.

HHHartmann, 2019-06-04:
Try using node.input() and node.ouput().

Alfrederson, 2019-06-04:
> Try using node.input() and node.ouput().

Good suggestion.

Node.input uses two externs, one is a lua_Load and the other is the same function I'm calling to force the interpreter to read from the spp queue, which is user_process_input. I'm only a bit concerned with that implementation, since any rework that for some reason got rid of gLoad and user_process_input would break that functionality.

Node.output stores stdout and calls redir_write instead when something is written to stdout (I'm assuming print commands write to stdout). I think I could also write a function that wrote data to the SPP to be sent to the bluetooth master, and at the same time wrote to uart0 by calling the old function. Do you think this could be done and at the same time leave the file open/read/write commands intact?

Alfrederson, 2019-06-10:
I tried doing the node.output and node.input way and guess what? It works. 

Running commands sent from the bluetooth interface is a simple matter of copying the messages into gLoad and calling user_proccess_input, and writing to the bluetooth interface is done by calling esp_spp_write.

There's still an issue, whereas the ESP32 will get stuck in an infinite loop if I call file.open() from the android cellphone, but not if I do it like:

 function fopen(filename,mode) return file.open(filename,mode) end
file = fopen("filename","w")

It also crashes with file:close().

Alfrederson, 2019-06-15:
Here it is
This is used by enabling the bluetooth controler in menu config, setting it as BTDM, disabling BLE and something else I don't remember, basically you need to intuitively set it as a BT Classic device

Then you drop this folder in the components directory (alongside driver_console, driver_can, driver_i2c, etc)

Add
#include "driver/btspp.h"
to base_nodemcu/user_main.c
and call btspp_init(); after start_lua() inside void app_main()

When the firmware boots up, the ESP32 will be visible with the "Slimy yet satisfying" devicename, it takes no PIN to pair (because I didn't implement GAP), and after your phone/tablet is paired to it you should be able to send lua commands over a Bluetooth Serial app.

[driver_btspp.tar.gz](https://github.com/nodemcu/nodemcu-firmware/files/3292552/driver_btspp.tar.gz)

closed 2020-06-23 by stale[bot]