nodemcu/nodemcu-firmware #2788: WS2812 not driving on ESP32

stuartpb, 2019-06-10:
### Expected behavior

Running a line like `ws2812.write{pin = 4, data=string.char(16):rep(24)}` should set 8 pixels to low brightness.

### Actual behavior

There's a brief flash of power out of the designated pin, but it doesn't appear to be timed correctly, because the WS2812 strip doesn't change at all. (This strip works fine when driven by an ESP8266.)

The strip is wired up to 3.3V for power, and I was able to see some activity somehow earlier today, but it all produced weird inconsistently-colored output like #2040 (ie. outputting a repeated byte like 64 led to a white pixel, a green pixel, a slightly-dimmer green pixel, and no further pixels).

This is all in debugging via terminal from a freshly-build firmware image. There's nothing else running on my chip.

### NodeMCU version

dev-esp32

### Hardware

ESP32-WROVER on a KE32-WROVER-C10 dev chip.

Console spam after restart:

```
I (30) boot: ESP-IDF v3.2-103-g4aa1058e8 2nd stage bootloader
I (30) boot: compile time 02:17:24
I (32) boot: Enabling RNG early entropy source...
I (36) boot: SPI Speed      : 40MHz
I (40) boot: SPI Mode       : DIO
I (44) boot: SPI Flash Size : 4MB
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00180000
I (81) boot:  3 spiffs           unknown          c2 00 00190000 00270000
I (89) boot: End of partition table
I (93) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x23fe8 (147432) map
I (153) esp_image: segment 1: paddr=0x00034010 vaddr=0x3ffb0000 size=0x01edc (  7900) load
I (156) esp_image: segment 2: paddr=0x00035ef4 vaddr=0x40080000 size=0x00400 (  1024) load
I (160) esp_image: segment 3: paddr=0x000362fc vaddr=0x40080400 size=0x09d14 ( 40212) load
I (184) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0xa4e4c (675404) map
I (419) esp_image: segment 5: paddr=0x000e4e6c vaddr=0x4008a114 size=0x0827c ( 33404) load
I (432) esp_image: segment 6: paddr=0x000ed0f0 vaddr=0x400c0000 size=0x00064 (   100) load
I (443) boot: Loaded app from partition at offset 0x10000
I (443) boot: Disabling RNG early entropy source...
I (444) cpu_start: Pro cpu up.
I (447) cpu_start: Starting app cpu, entry point is 0x400812ac
I (0) cpu_start: App cpu up.
I (458) heap_init: Initializing. RAM available for dynamic allocation:
I (465) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (471) heap_init: At 3FFB7FF0 len 00028010 (160 KiB): DRAM
I (477) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (483) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (490) heap_init: At 40092390 len 0000DC70 (55 KiB): IRAM
I (496) cpu_start: Pro cpu start user code
I (178) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.

Mounting flash filesystem...
I (580) wifi: wifi driver task: 3ffc1170, prio:23, stack:3584, core=0
I (580) wifi: wifi firmware version: 2af77cc
I (580) wifi: config NVS flash: enabled
I (580) wifi: config nano formating: disabled
I (580) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (590) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (630) wifi: Init dynamic tx buffer num: 32
I (630) wifi: Init data frame dynamic rx buffer num: 32
I (630) wifi: Init management frame dynamic rx buffer num: 32
I (630) wifi: Init management short buffer num: 32
I (630) wifi: Init static rx buffer size: 1600
I (640) wifi: Init static rx buffer num: 10
I (640) wifi: Init dynamic rx buffer num: 32
NodeMCU ESP32 build unspecified powered by Lua 5.1.4
lua: cannot open init.lua
```

devsaurus, 2019-06-10:
If you suspect incorrect timing and assume the esp8266 driver as a reference, then you could try to adapt T0H etc. timings at https://github.com/nodemcu/nodemcu-firmware/blob/e30904b60760f8c290ada6deb2a44480ab884ba2/components/platform/ws2812.c#L40

The esp8266 encodes with 1x 312ns and 3x 312ns, which translates into 3 and 9 in esp32 context.

stuartpb, 2019-06-11:
I took a crack at adjusting the timings, but I didn't seeing any change in behavior. I don't know if it's necessarily a timing issue - seeing as how other folks had said it works for them, I was kind of looking to see if someone else would check to see if the problem reproduces for them (ie. something else broke WS2812 and nobody's noticed).

Here's a curious thing: I hooked up a plain old LED to my data-signalling line when testing the ESP32, to ensure I wasn't accidentally sending commands from the wrong pin. When I'd run a `ws2812.send`, I'd see the LED flash a little (though no activity on the WS2812 strip).

I've swapped the configuration back to my ESP8266 setup, where data is being sent to the strip constantly, and I'm not seeing *any* activity visible on my LED trace. Maybe the signals from the ESP32 being slow/high enough to be visible like this is a sign that something's up?

devsaurus, 2019-06-12:
Well, with a 3v3 supply the ws2812 strip is operated out of spec afaict. Results might be not consistent.
Best would be to supply with 5v and feed the signal from esp32/esp8266 through a level shifter.

stuartpb, 2019-06-19:
No, it's not that: I've got the chip hooked up to 5V and am using [the logic level shifter recommended for use with WS2812s by Adafruit](https://www.adafruit.com/product/1787), and I'm still seeing the same behavior.

stuartpb, 2019-06-21:
Is there some kind of known-good test file / image I could run to see if it's my code, versus my build of the firmware, versus my chip?

devsaurus, 2019-06-21:
This is my std testcase for 8-RGB stripe (derived from https://nodemcu.readthedocs.io/en/dev-esp32/modules/ws2812/#example_1)
```lua
ws2812.write({pin = 4, data = string.char(0,32,0):rep(8)})
i, buffer = 0, ws2812.newBuffer(8, 3); buffer:fill(0, 0, 0); tmr.create():alarm(50, 1, function()
  i = i+1
  buffer:fade(2)
  buffer:set(i % buffer:size() + 1, 0, 0, 32)
  ws2812.write({pin = 4, data = buffer})
end)
```

closed 2020-06-30 by stale[bot]