nodemcu/nodemcu-firmware #2792: Improve error handling for: "PANIC: unprotected error in call to Lua API (attempt to concatenate a nil value)"

maciejmiklas, 2019-06-14:
Sometimes my program crashes with such error:
`PANIC: unprotected error in call to Lua API (attempt to concatenate a nil value)`

Since it's not a small application finding a reason for it is really hard. It would be nice, if we could get some more info in such situation (file, line), or at least node would restart.

I'm on ESP32 latest Frimware

TerryE, 2019-06-15:
Maciej, I am currently writing this out with my Lua 5.3 and to simplify the migration, I am back-porting some of these fixes to ESP8266 Lua 5.1.  I am not sure whether we'll get around to doing the backport to the ESP32. If not then this will need to wait to the common 5.3 release. Sorry.

maciejmiklas, 2019-06-15:
super and thank you :)

TerryE, 2019-11-09:
@maciejmiklas, Maciej, BTW in general if you want robust applications then it is sensible to do protected calls using `pcall()` or `xpcall()` and then you won't get panics anyway.

maciejmiklas, 2019-11-09:
I have a few timers each one executes method trough `pcall()`, but I'm still getting sometimes panic. It's hard to determine the cause for it.

closed 2020-11-21 by stale[bot]