nodemcu/nodemcu-firmware #2794: luac.cross segfault when given -l option

jasaw, 2019-06-18:
### Expected behavior

```
./luac.cross -o luac.out lua_examples/lfs/_init.lua
./luac.cross -l luac.out
```
List the compiled instructions etc.

### Actual behavior

Dereferences a NULL pointer and segfaults.

### Test code

See expected behaviour section.

### NodeMCU version
Dev branch, commit 1990f95740e9f7d375b1157a25d0b6b508586c27

### Hardware
Any Linux machine.

HHHartmann, 2019-06-18:
As far as I can recall the -l had to be used on the same command. 

```./luac.cross -l -o luac.out lua_examples/lfs/_init.lua```

jasaw, 2019-06-18:
`./luac.cross -l -o luac.out lua_examples/lfs/_init.lua` does work.

Question is, why does `./luac.cross -l luac.out` (separate commands) work on some files, but not `_init.lua`?

Anyhow, if -l option has to be used on the same command, it should probably check for lua source files from command line arguments before proceeding?

TerryE, 2019-06-18:
```
./luac.cross -o luac.out lua_examples/lfs/_init.lua
```
actaully creates an LC file albeit with the name `luac.out`, and compiling this segfault because of #2727.  So this is a duplicate.

closed 2019-06-18 by TerryE