nodemcu/nodemcu-firmware #2796: EP32: Ethernet support

devsaurus, 2019-06-18:
### Missing feature
Support for ESP32's ethernet API.

### Justification
There is a growing amount of boards combining the ESP32 with an Ethernet phy chip.

### Workarounds
None.

### Description
We've seen a couple of requests for Ethernet support on ESP32 like #2134 and #2525 @Lir10 @fengmingfm @heythisisnate. I've taken the time to investigate this topic with a prototype `eth` module implementation and found that it's quite straight forward with esp-idf.

Before preparing a PR, I'd like to understand the boundary conditions for an `eth` module. Up to now I see
* support for different phy chips (idf supports tlk110, lan8720, ip101)
* configurable phy address
* board specific wiring, e.g. clocking, enabling power
* maybe speed and duplex settings

Main objective of `eth` module is to prepare the hardware for upper layers like `net`, `mqtt`, etc. I assume that sending and receiving Ethernet frames is out of scope for a Lua module.

Any further thoughts?

heythisisnate, 2019-06-20:
I'm very excited about this! Sorry I don't have much to add to the technical discussion but I'd be happy to provide support, testing, and hardware to help move along this effort.

devsaurus, 2019-06-25:
Module code is at https://github.com/devsaurus/nodemcu-firmware/tree/esp32-eth, needs more testing and full docs.

Example code for ESP32-GATEWAY:
```lua
function ev(event, info)
    print("event", event)
    if event == "got_ip" then
        print("ip:"..info.ip..", nm:"..info.netmask..", gw:"..info.gw)
    elseif event == "connected" then
        print("speed:", eth.get_speed())
        print("mac:", eth.get_mac())
    end
end

eth.on("connected", ev)
eth.on("disconnected", ev)
eth.on("start", ev)
eth.on("stop", ev)
eth.on("got_ip", ev)
eth.init({phy  = eth.PHY_LAN8720,
          addr = 0,
          clock_mode = eth.CLOCK_GPIO17_OUT,
          power = 5,
          mdc   = 23,
          mdio  = 18})
```

AddoSolutions, 2019-07-01:
@devsaurus Fantastic work, thanks for putting that together!

Looking as a noobie here, what are next steps to this? I could help with docs for sure. And for testing, do we need unit tests or just on-board tests?

Thanks!

devsaurus, 2019-07-01:
@AddoSolutions mainly tests to reproduce correct initialization and operation on other boards than the ESP32-GATEWAY. I'm currently waiting for shipment of another type - if you can test drive the code and report results that would be very much appreciated.
Docs will are pretty straight forward.

PR #2820 (2019-07-20): Esp32 eth
Fixes #2796

- [x] This PR is for the `dev-esp32` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

This PR adds the `eth` module based on idf's Ethernet layer.
I've tested the module with an ESP32-GATEWAY board. Other boards with one of the supported PHY chips should work unless they require a specialized in [... cut]
closed 2019-07-20 by marcelstoer