nodemcu/nodemcu-firmware #2798: Document the calculation of the load address of a SPIFFS image.

cwrseck, 2019-06-19:
### Missing feature

Using an Adafruite HUZZAH, and nodemcu-firmware-20190405

Hitherto I've not loaded a SPIFFS file with a firmware update,
since adding about 7 kB of files to local/fs and running make
blows up with the message "Not enough space: fatal error"

(In fact, this error probably comes from spiffsimage and
simply refers to the build for the 512 kB part.  A more
informative message, with the name of the program and
the specific error would be useful; or possibly matching
messages for successful image builds.)

Since I (thought I) had no SPIFFS images I tried to find the
loading address by trial and error, adding the size of the
NodeMCU firmware file to the firmware load address.  This
produced the value 0x83000, but loading the SPIFFS image
at that address bricked the ESP8266 with:
```
> file.format()
*** ERROR ***: unable to format. FS might be compromised.
It is advised to re-flash the NodeMCU image.
```
Reflashing the NodeMCU image produced no improvement, until
my investigation of the "fatal error" message showed that
a SPIFFS image for a 4MB part had been built, complete with
an address of 0x90000.  Reflashing that image unbricked the
ESP8266.

I have no idea where the 0x90000 address came from. The derivation
of the SPIFFS load address badly needs further documentation,
which would make the use of spiffsimg a lot safer.

### Justification
Make the use of spiffsimg a lot safer

### Workarounds
Building firmware with files in local/fs and using the address from the names of the binaries.

C W Rose

jmattsson, 2019-06-26:
Have you looked at `file.fscfg()`? It provides the address and size of the spiffs area.

cwrseck, 2019-06-26:
Yes, with a working ESP8266 you can use fscfg() (provided that the SPIFFS file system is always loaded in the same place, which I'm pretty sure it is).  However, without the hardware, as I was, you are kind of stuck.  fscfg() must get its information from somewhere - it might be worth documenting exactly where.

C W Rose

jmattsson, 2019-06-27:
@pjsg I think you're the one most across the spiffs positioning at the moment - is there a comprehensible way of explaining the aligning, offsetting, magic-value-searching etc that happens when trying to find the spiffs?

pjsg, 2019-06-27:
As I recall, there are two main possibilities: either you define SPIFFS_FIXED_LOCATION or you don't. If you define SPIFFS_FIXED_LOCATION, then the spiffs image will be placed there, and the firmware will look for it there.

If you don't specify it, and you are building a spiffs image from the makefile, then it will be placed fairly close to the end of the firmware. 

In both cases, the resulting image file will be named something like:

0x90000-32mb.img

I.e. this is the image to be flashed at offset 0x90000 for a 32mb chip (4MB). 

If you *don't* flash a spiffs image, then the firmware attempts to find an existing image in the flash. The rules for this are quite complex, but the basic idea is to check all likely starting locations within a range of the end of the firmware. If the firmware increases in size by too much, then it will overwrite the start of the spiffs image, and so it won't find a valid image.

If the firmware can't find a valid spiffs filesystem, then it tries to format a new one. It picks a location (either ~64k from the end of the firmware or the fixed location) and then formats it. The length is either the fixed length configured, or up to the end of the available area.

The detection of spiffs file systems requires that the start and the end of the filesystem have not been modified. There is the case that having a filesystem (say) running from 1MB to 3.5MB could be overwritten by one running from 2MB to 3MB and this wouldn't (as I recall) be detected. In the worst case, just erase your flash chip and reflash everything.

Does this help?

cwrseck, 2019-06-27:
Hmm.  I suspect that this problem reaches the "all too difficult" level.  Perhaps all that can be done is to hightlight somewhere early in the installation docs that you need to put files in local/fs and record the load address of the resulting image (from bin), even if you subsequently remove the files and rebuild.

The alternative would be to write a specific program to match the current SPIFFS build and write out the load address; updating the documentation seems easier.

C W Rose

TerryE, 2019-06-27:
With the current `dev` version we provide a tool `nodemcu-partitions.py` which lists your current ESP8266 partition table -- that is including where your default LFS and SPIFFS partitions are located, and this allows you to move / resize SPIFFS partition and to load a new SPIFFS image.  Ditto LFS.

Note that since the runtime of many SPIFFS operations is directly proportional to the size of the FS, in my view it is a good idea to keep the SPIFFS partition as small as you safely require.

cwrseck, 2019-06-29:
@TerryE  Yes, well, that was why I was messing with SPIFFS in the first place; pulling it down to 128K
However, it looks as if the move to a partition table will fix the problem.

C W Rose

TerryE, 2019-11-11:
See #2960

closed 2019-11-11 by TerryE