nodemcu/nodemcu-firmware #2800: SPIFFS_OPEN causing a reset when called from input not generated by uart?

Alfrederson, 2019-06-20:
### Expected behavior
file.open(filename,mode) should return a file handle when called from a custom task
### Actual behavior
It resets the ESP32
### Test code
I have a callback for SPP.
When this callback is called for a ESP_SPP_DATA_IND_EVT, it puts the received bytes onto a queue, and counts how many bytes it received.
When it receives a newline, it puts the amount of bytes it received on another queue and sets a byte inside an event group indicating it has a complete line (which from what I've read constitutes a complete Lua chunk).
I have a task that waits until that bit is set for it to read how many characters it needs to read from a queue (_spp_char_count_queue), an then it reads cmd_length characters from _spp_rx_queue.
It then inserts that chunk into the global gLoad and calls user_process_input.
This works fine for defining variables, running loops, changing GPIO states and even reading from a file if it was open by a line sent through UART, but when it tries to call file.open(), the ESP32 resets with code rst:0x3 (SW_RESET).
I spent a lot of time trying to find exaclty where the crash happens but I haven't gone any deeper than SPIFFS_OPEN. Could this be a call stack depth issue, since file.open becomes lua_dojob -> file_open -> vfs_open -> myspiffs_realm -> myspiffs_vfs_open -> SPIFFS_OPEN -> spiffs_object_find_object_index_header_by_name -> etc ?

```C
extern lua_Load gLoad;
void btspp_input_inject_task(void * arg){
    // task param t is an int pointer but I'll use it as a simple int
    // to count how many itens I'll take out of the queue
    for(;;){
        if(xEventGroupWaitBits(_spp_event_group,SPP_NEED_INJECT,pdTRUE,pdFALSE,portMAX_DELAY)){
            // check the length
            uint16_t cmd_length;
            uint8_t* buf;
            int i;
            cmd_length=0;
            xQueueReceive(_spp_char_count_queue,&cmd_length,0);

            buf = (uint8_t*)malloc(cmd_length);
            memset(buf,0,cmd_length);

            for(i=0;i<cmd_length;i++){
                if(xQueueReceive(_spp_rx_queue,&buf[i],0)==pdTRUE){
                    //
                }else{
                    ESP_LOGI(TAG,"Reading past the queue!");
                    break;
                }
            }

            // interpret the input line
            buf[cmd_length-2]=0;
            lua_Load *load = &gLoad;
            // this shouldn't lock up the esp32
            while( load->line_position != 0 );
            memcpy(load->line,buf,cmd_length-1);
            load->line_position = strlen(load->line)+1;
            load->done=1;
            free(buf);
            user_process_input(true);
        }
    };
    vTaskDelete(NULL);
    _spp_inject_task_handle=NULL;
}
```
### NodeMCU version
This is pertaining the ESP32 branch.

### Hardware
ESP32-wroom module.

8<------------------------ END BUG REPORT -------------------------------------

jmattsson, 2019-06-26:
My initial thoughts here is that you might end up starving the idle task enough that it doesn't feed the watchdog. Instead of hard-looping in that `while`, try calling `taskYIELD()` to actually give the Lua VM task a chance to process its existing input.

Alfrederson, 2019-06-26:
> My initial thoughts here is that you might end up starving the idle task enough that it doesn't feed the watchdog. Instead of hard-looping in that `while`, try calling `taskYIELD()` to actually give the Lua VM task a chance to process its existing input.
That one could be a problem if at the exact moment that line is running the interpreter is already parsing something etc. but there's something special about the vfs_open "chain". If I disable the watchdog and call file.open() through this task, the firmware enters an infinite loop outputing repeated characters, which could be the beginning of an error message that is never finished or something like that.

jmattsson, 2019-06-27:
Hmm, the `vfs_open` stuff goes deep into the IDF itself. When you hit the crash, do you get a stack trace printed? If you use `make monitor` that will automatically decode the addresses into function names when it sees the stack trace line. If you're not getting it, maybe try building for single-core (components -> FreeRTS -> Run FreeRTOS only on first core) to force more contention and hopefully get a stack trace (you'll need th` taskYIELD()` version in this case, no doubt). You can also enable the gdb stub (component -> ESP32-specific -> Panic handler behaviour), at which point `make monitor` will drop you straight into gdb in case of a crash.

Alfrederson, 2019-06-29:
@jmattsson I was getting one before I changed some options, I'll try reenabling it. I almost forgot ESP had a stack trace decoder.

Alfrederson, 2019-07-05:
I enabled gdbstub and ran make monitor, when I get the infinite loop and wdt reset, I get this

`rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:5284
load:0x40078000,len:9768
load:0x40080400,len:6328
entry 0x40080710
W (172) boot: PRO CPU has been reset by WDT.
W (172) boot: WDT reset info: PRO CPU PC=0x40092619
0x40092619: panicPutChar at /home/godofredo/nodemcu-firmware-esp3/sdk/esp32-esp-idf/components/esp32/panic.c:707

W (172) boot: WDT reset info: APP CPU PC=0xa333eded
`

Could panicPutChar be the culprit?

Alfrederson, 2019-07-06:
Putting the received characters in uart0Q gives the same result. This does not make any sense, considering uart0_rx_intr_handler in console.c does pretty much the same thing.

jmattsson, 2019-07-08:
Ah, wait! You're calling `user_process_input()` straight from your own RTOS task. That means you're effectively ending up with two separate threads touching the LVM at the same time, which is a Bad Thing(tm) resulting in all manner of undefined behaviours. What you'll need to do is to post an event into the LVM thread (see `components/task/include/task.h`), and then do the pulling-off-the-queue + load within the LVM task/thread. Effectively, once `xEventGroupWaitBits()` returns in your RTOS task, you'd `task_post_medium()` to notify the LVM thread, and that would then invoke a your handler from within the LVM thread/RTOS-task, where you do the read from the queue and process it.

Apologies for being slow and not twigging to this way earlier.

Alfrederson, 2019-07-08:
But then why would I get this specific undefined behavior only when calling file functions? Like, if the file is already there, opening it doesn't cause an infinite loop, but I get one if I open it and then do file.close(). user_process_input() simply posts a task too, so I get the same result if I use it or if I use task_post_low(handle_input) etc.

jmattsson, 2019-07-09:
Yours truly was sure `user_process_input()` was a run-to-completion function, not a task-poster. D'oh! Clearly I had my wires cross in my memory with the 8266 version.

That said, the access to the `gLoad` from any other thread than the LVM thread is still an issue.

And I'm not discounting that there is something specific in the `file.open()/close()` path that's an issue, I'm just trying to get the obvious things out of the way first, to hopefully make it easier to pinpoint the remaining one! I don't suppose you've managed to reproduce it purely within the LVM task, have you?

Btw, do you have `CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU0=y` and `CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU1=y` set in your config? The reset output above looks different to what I'm used to seeing on an idle-starvation timeout.

Alfrederson, 2019-07-13:
It's enabled, it doesn't seem to give any helpful information

jmattsson, 2019-07-19:
What I tend to do is put a "landmine" in the call path to force a crash and exit to gdb at that point. A simple `*(int *)0 = 0;` gives a guaranteed null pointer dereference and crash at that point. If necessary, I put it behind an `if (crash_enabled)`, and then just set that flag from a Lua module function (`node.heap()` being the typical function I drop such debugging into). That way I can get through whatever startup code that might share that callpath, then `node.heap()` to prime the mine, and then trigger the actual failure scenario.

If you're set up for JTAG, have a look at https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/app_trace.html; that might be of use too. I have yet to hook up my flyswatter2, so I haven't tried it myself yet.

o07sai, 2019-07-24:
I also encounter this problem. After many experiments, it is found that if BLE is enabled and file module methods are called, such as open, write, read, list, reset will be triggered. Current version: 511bc01dce8f947729e2c93551ee7d44af7e5175. But yesterday, I used a new version and the problem did not arise. The logs at the time of the problem are as follows:
lua: cannot open init.lua

Heap size::55008.
=node.heap()
=node.heap()
56008
_dir=function() local k,v,l print("~~~File ".."list START~~~") for k,v in pairs(file.list()) do l = string.format("%-15s",k) print(l.." : "..v.." bytes") end print("~~~File ".."list END~~~") end _dir() _dir=nil
_dir=function() local k,v,l print("~~~File ".."list START~~~") for k,v in pairs(file.list()) do l = string.format("%-15s",k) print(l.." : "..v.." bytes") end print("~~~File ".."list END~~~") end _dir() _dir=nil

rc522.lua       : 15447 bytes
test            : 5 bytes
test100.lua     : 3301 bytes
test4.txt       : 9593 bytes
btpp.lua        : 418 bytes
test.bmp        : 276534 bytes
bmp.lua         : 1140 bytes
~~~File list END~~~
> r,u,t=file.fsinfo() print("Total : "..t.." bytes\r\nUsed  : "..u.." bytes\r\nRemain: "..r.." bytes\r\n") r=nil u=nil t=nil
r,u,t=file.fsinfo() print("Total : "..t.." bytes\r\nUsed  : "..u.." bytes\r\nRemain: "..r.." bytes\r\n") r=nil u=nil t=nil
total: 2412361, used:312746
Total : 2412361 bytes
Used  : 312746 bytes
Remain: 2099615 bytes

> =node.heap()
=node.heap()
55384
> bthci.adv.setdata(encoder.fromHex("080861626364656667"), function(err) print(err or "Ok!") end)
bthci.adv.setdata(encoder.fromHex("080861626364656667"), function(err) print(err or "Ok!") end)
> Ok!
bthci.adv.enable(1, function(err) print(err or "Ok!") end)
bthci.adv.enable(1, function(err) print(err or "Ok!") end)
> Ok!
bthci.adv.enable(1, function(err) print(err or "Ok!") end)
bthci.adv.enable(1, function(err) print(err or "Ok!") end)
> Ok!
 _dir=function() local k,v,l print("~~~File ".."list START~~~") for k,v in pairs(file.list()) do l = string.format("%-15s",k) print(l.." : "..v.." bytes") end print("~~~File ".."list END~~~") end _dir() _dir=nil
 _dir=function() local k,v,l print("~~~File ".."list START~~~") for k,v in pairs(file.list()) do l = string.format("%-15s",k) print(l.." : "..v.." bytes") end print("~~~File ".."list END~~~") end _dir() _dir=nil
~~~File list START~~~
rc522.lua       : 15447 bytes
test            : 5 bytes
test100.lua     : 3301 bytes
test4.txt       : 9593 bytes
btpp.lua        : 418 bytes
test.bmp        : 276534 bytes
bmp.lua         : 1140 bytes
~~~File list END~~~
> r,u,t=file.fsinfo() print("Total : "..t.." bytes\r\nUsed  : "..u.." bytes\r\nRemain: "..r.." bytes\r\n") r=nil u=nil t=nil
r,u,t=file.fsinfo() print("Total : "..t.." bytes\r\nUsed  : "..u.." bytes\r\nRemain: "..r.." bytes\r\n") r=nil u=nil t=nil
total: 2412361, used:312746
Total : 2412361 bytes
Used  : 312746 bytes
Remain: 2099615 bytes

>  _dir=function() local k,v,l print("~~~File ".."list START~~~") for k,v in pairs(file.list()) do l = string.format("%-15s",k) print(l.." : "..v.." bytes") end print("~~~File ".."list END~~~") end _dir() _dir=nil
 _dir=function() local k,v,l print("~~~File ".."list START~~~") for k,v in pairs(file.list()) do l = string.format("%-15s",k) print(l.." : "..v.." bytes") end print("~~~File ".."list END~~~") end _dir() _dir=nil
~~~File list START~~~
rc522.lua       : 15447 bytes
test            : 5 bytes
test100.lua     : 3301 bytes
test4.txt       : 9593 bytes
btpp.lua        : 418 bytes
test.bmp        : 276534 bytes
bmp.lua         : 1140 bytes
~~~File list END~~~
> r,u,t=file.fsinfo() print("Total : "..t.." bytes\r\nUsed  : "..u.." bytes\r\nRemain: "..r.." bytes\r\n") r=nil u=nil t=nil
r,u,t=file.fsinfo() print("Total : "..t.." bytes\r\nUsed  : "..u.." bytes\r\nRemain: "..r.." bytes\r\n") r=nil u=nil t=nil
total: 2412361, used:312746
Total : 2412361 bytes
[... 93 lines cut ...]
�[0;32mI (28) esp_apptrace: Initialized TRAX on CPU1�[0m
�[0;32mI (0) cpu_start: Starting scheduler on APP CPU.�[0m

Mounting flash filesystem...
mount res: 0, 0
Task task_lua started.
�[0;32mI (725) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE�[0m
�[0;32mI (725) BTDM_INIT: BT controller compile version [74c9f57]�[0m
�[0;32mI (735) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE�[0m
�[0;32mI (805) phy: phy_version: 4008, 544f89f, Jan 24 2019, 14:54:06, 0, 0�[0m
I (1135) wifi: wifi driver task: 3ffe3b50, prio:23, stack:3584, core=1
I (1135) wifi: wifi firmware version: 2af77cc
I (1145) wifi: config NVS flash: enabled
I (1145) wifi: config nano formating: disabled
�[0;32mI (1145) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE�[0m
�[0;32mI (1155) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE�[0m
I (1195) wifi: Init dynamic tx buffer num: 32
I (1195) wifi: Init data frame dynamic rx buffer num: 32
I (1195) wifi: Init management frame dynamic rx buffer num: 32
I (1205) wifi: Init management short buffer num: 32
I (1205) wifi: Init static rx buffer size: 1600
I (1215) wifi: Init static rx buffer num: 10
I (1215) wifi: Init dynamic rx buffer num: 32
NodeMCU ESP32 build unspecified powered by Lua 5.1.4
lua: cannot open init.lua

Alfrederson, 2019-08-02:
@o07sai  so you think compiling nodemcu against a newer version of ESP-IDF should fix this? BTW, I get the same infinite string of "к?к?к?к?к?к?к?к?к?к?к?..." you did

o07sai, 2019-08-05:
@Alfrederson on NodeMCU dev-esp32 commit 7cb61a27d20fef3b868e95817089ee71b5b753fc, the problem seems like fixed. You can try it. And my email: o07sai@163.com.

Alfrederson, 2019-08-14:
> @Alfrederson on NodeMCU dev-esp32 commit [7cb61a2](https://github.com/nodemcu/nodemcu-firmware/commit/7cb61a27d20fef3b868e95817089ee71b5b753fc), the problem seems like fixed. You can try it. And my email: [o07sai@163.com](mailto:o07sai@163.com).

This is with a newer ESP-IDF or with the one that comes in the repository?