nodemcu/nodemcu-firmware #2806: U8g2: display drivers are not compiled

shimosaurus, 2019-06-21:
### Expected behavior
```
> print(u8g2.ssd1306_i2c_128x64_noname)
lightfunction: 0x402630d0
```
### Actual behavior
```
> print(u8g2.ssd1306_i2c_128x64_noname)
nil
```
### NodeMCU version
Actual `dev` branch.
Broken in PR: [Updated ROTables 1st tranch #2742](https://github.com/nodemcu/nodemcu-firmware/pull/2742) 

### Solution 
Remove "#" from "#binding" in `/app/modules/u8g2.c`

patch:
```
@@ -807,7 +807,7 @@
 
 #undef U8G2_FONT_TABLE_ENTRY
 #undef U8G2_DISPLAY_TABLE_ENTRY
-#define U8G2_DISPLAY_TABLE_ENTRY(function, binding) LROT_FUNCENTRY(#binding,l ## binding)
+#define U8G2_DISPLAY_TABLE_ENTRY(function, binding) LROT_FUNCENTRY(binding,l ## binding)
 LROT_BEGIN(lu8g2)
   U8G2_DISPLAY_TABLE_I2C
   U8G2_DISPLAY_TABLE_SPI
```

TerryE, 2019-06-22:
Thanks  @shimosaurus, I will sneak this into my next PR. Sorry about this, and the fix is much appreciated. 😊

marcelstoer, 2019-06-22:
> I will sneak this into my next PR

No need, I just fixed this on `dev`.

commit 2d584a2a (2019-06-22): Fix ROTable call

Fixes #2806
closed 2019-06-22 by marcelstoer