nodemcu/nodemcu-firmware #2811: RFC Why have different coding standards for ESP32 and EP8266 modules?

TerryE, 2019-06-23:
This is really one where I am looking for views from Arnim and Johny.  As you should be aware, I am going through the process of preparing for the Lua53 migration, and one of my main derisking strategies is that I an _not_ going to add new `dev` and `master`  branches for Lua 5.3 because everything apart from the `lua` and `lua53` sub-components will be common.

I am also review how this will carry across to porting Lua53 to the ESP32 environment, and really find myself asking the Q:

- _Why are the coding standard / APIs different for the two variant platforms?_

I can see that it was easier just to fork and patch when working to a deadline.  I can also understand why modules might have to be different when they depend on different hardware features.  With a little bit of effort we could have aligned the coding standards and used a little bit of define magic to hide virtually all of the API and coding differences.

The fork and patch lowers the upfront effort, but though-life it just causes the project to bleed because we are having to divide or limited resources between the two code variants: there is a conversion overhead every time we have to migrate a module across platforms, and we can't just cherry-pick fixes from one platform to the other.

For example:
-  ESP32 code follows the `<stdint.h>` conventions for the various integer definitions (`uint32_t` etc.) wheres the ESP8266 code follows the SDK "c_types.h" conventions (`uint32` _without the `_t` suffix_)
- ESP32 code follows the `<sting.h>` conventions for the string and memory operation (`stcpy()` etc.) whereas the ESP8266 code again follows the SDK convention of using the `c_` prefix.
-  There are other minor API changes that have been made in ESP32.

Let me be clear: I support the choice of using POSIX-compliant + C99 coding standards whenever practical, so I don't disagree with what you've done for the `dev-esp32` branch.  However, my assertion is the it would be better in the long run if all of code including that for the ESP8266 adopted this strategy.
    
I had to solve most of these issues in order to build the cross compiler on the host platform.  Surely, it is worth the effort to converge all of this?

As a first step, I can take on the job of reworking the modules.  I also suggest that 
-  I rename `app` to `components`.
-  We spilt the `modules` into three sub-directories:
   -  **core_modules**. Those modules that are in essence mandatory for any firmware build and their omission isn't really supported.
   -  **common_modules**.  Those modules that should (in principle) work on both platforms, and where any platform-specific variations are handled through `#ifdef` logic.
   -  **esp8266_modules**.  Those modules that have ESP8266 architectural dependencies that means that the ESP32 implementation must be different.

Our goal should be that the source for `components/lua53`, `components/core_modules` and `components/common_modules` should be the same (subject possibly to cherry pick delays) for the two platforms.  Am I being unrealistic here?

If we go this way the perhaps the `esp8266_modules` list might be long initially, but we should move as policy to refusing to accept new modules or major rework of modules for either platform unless can be migrated into `common_modules`.

We should also ensure that any Lua API extensions are also common to both platforms -- which is going to have to happen anyway if `components/lua53` is to be shared.

jmattsson, 2019-06-23:
I, with a firey passion, hate the `c_xxx.h` headers. They conflict horribly with the standard C headers, and serve no good purpose these days. I got rid of them once already when I ported the 8266 to RTOS (which we never really adopted due to the RAM usage of RTOS at the time - I don't know if that situation has changed). I'd like to have time to get rid of them from the non-os SDK too, and replace them with stubs in `sdk-overrides` which simply include the correct C header.

It would probably be a great idea to spike out and see whether the RTOS SDK would be usable for us on the 8266, since it should be *far* easier to maintain support for both chips under a common framework (the 8266 RTOS is IDF these days). I totally do not have the time for that at the moment though. Most of the necessary code changes could probably be lifted from my earlier work (still accessible at https://github.com/DiUS/nodemcu-firmware/tree/dev-rtos), if anyone else wants to take a stab at it.

As for modules, yes, I agree in principle, though there will definitely be modules which are ESP32-only as well, given the hardware there is far more powerful than the 8266.

We'll certainly have a ways to go with the `platform` layer; though at least with partition support on the 8266 it should be easier now than it was last time I poked at it.

TerryE, 2019-06-23:
We take to much of a hit on RAM with RTOS on the ESP8266 -- we had negative comments about free RAM just going to SDK 3.0.

So long as we can get as much of the source as possible shared, then I think we can live with RTOS vs SDK.

jmattsson, 2019-06-24:
We need to spruik LFS more then! :D

devsaurus, 2019-06-25:
I can't add anything to the `c_xxx` headers topic. My impression is that we should unify towards the standard C headers, if feasible.

Regarding the partitioning into `core_modules`, `common_modules`, and `esp8266_modules` I also think this is a reasonable approach. I guess this implies that from `dev-esp32` side we want a `esp32_modules`. Would also enable us to gradually converge the two platforms.
E.g. highest prio to have a unique `core_modules` and a `common_modules` that contains low-hanging fruits first. While keeping more complex modules in a backlog and converge them over time from `esp8266_modules` and `esp32_modules` into `common_modules`. Just a thought to reduce the amount of mandatory efforts before going live.

For the `platform` layer I agree to the goal of a common API unless we accept to propagate platform specific items to Lua land.
While working on the esp32 code base I got the impression that `platform` is used so far only when common code is shared between modules. In my opinion we could generate benefits from moving functionality into `platform` that's even used in a single module. E.g. the ws2812 low-level pulse train generation is specific to the respective Lua module but has completely different implementations for the two architectures. This Lua module could be moved to `common_modules` if we achieve abstraction on `platform` layer.

TerryE, 2019-06-26:
The `platform` abstraction layer was introduced by the `elua` project largely to isolate the Lua core from the various targets it supported, so it was carried into the first nodemcu versions. However, this seems to have been more of a more of a pragmatic hack than properly designed.  As a result the layer wraps some stuff that doesn't need to be wrapped and doesn't wrap stuff that should be. 

For a sensible `lua53` and `common_modules` implementation across both platforms, then these code files should either be using POSIX-consistent API calls or going through a unified platform layer and we should only make direct calls to ESP ROTS / SDK services if and where Espressif has ensured that their relevant APIs are common to both platform variants.

What I mean by API consistent is that the subset of the string library used in SDK applications for example compile error-free on the ESP8266, and because our `ld/defsym.ld`remaps the `strcmp` reference to `ets_strcmp`, this link to the correct image.  So we should just use `<string.h>` in all of our source code and avoid tattooing our code with unnecessary non-standard API usage.

An example of where the service APIs could be common are the platform `flash_api.h` variants which are different for no clear reason.  IMO, we should be using the same header even if we some conditional declarations and conditional includes of the respective Espressif RTOS/SDK headers and that way any of our code which calls the SPI services can at least be common across both platforms at a source level.

There are other anomalies, and I can use an example that I introduced: our tasking API.  Even though the SDK is entirely event-driven and the RTOS application (like node.js) implements the event driven framework within the application, our tasking interface is effectively the same at an API level and is actually a core service they both Lua VMs depend on.  It should therefore be a proper platform service with a common API.  So with my next preparation patch I will be retiring the `task.c` library and moving these functions into platform services and this will allow me to initialise the task interface as part of `platform_init()`.  I will keep a `task.h` header in the intermediate term for backwards compatibility, but this will simply remap the task API and defines onto the new platform ones.

TerryE, 2019-07-21:
We are suffering from entropy: it takes effort that we haven't been putting in to keep things standard.  When we do an effective fork as is the case with dev <-> dev-esp32 we just get even more disorder. As Johny has mentioned in #2838, the sdk lib is not really compatible with newlib so trying to share modules between the SDk and RTOS/newlib based works is going to take a lot of effort.  

For example, we nominally have a platform abstraction layer but this isn't really enforced.  So **39** out of the **72** C files in the esp8266 `app/modules` directory include SDK headers and hence (I assume) are using SDK-specific types, structures and defines.  Few if any of these apply to esp32 code. So as JM and AL know, migrating modules takes work.

If we are going to harmonise the module code base downstream then as @jmattsson has said on many occasions: this would be a _lot_ easier if both implementations ran on RTOS and all that takes is for the ESP8266 to suddenly have an extra 32Kb RAM to Lua and RTOS can fit together -- and that isn't going to happen; is it?

Take something as simple as uint32's, we have at least 5 type versions for this `uint32_t`, `uint32`, `lu_int32`, `unsigned` and `unsigned int`.  The most common use is the first which is declared in `c_types.h` on the SDK builds and in `<stdint,.h>` in IDF builds.  You can't include both in a source fill since the definitions clash.

Needs more thought.

jmattsson, 2019-07-22:
Stand by for PR evicting `c_types.h` too. That's the first step in tidying up our numeric types.

TerryE, 2019-07-29:
Now that I have a baseline of the `app/modules` directory compiling clean with lua53, I think that we can close this issue.

closed 2019-07-29 by TerryE